nasa/fprime #1685: OSAL Cleanup

LeStarch, 2022-09-30:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The Os package (out OSAL layer) is a mess.  It is effectively two packages: libOs.a and libOsImplementaion.a .... but they are munged into a super lib.  We should cleanly separate the two such that implementers can swap out cleanly their own Os integration.

We also need to cleanup the distinction between full-Posix, Linux, and Darwin and make those distinctions clear.

ThibFrgsGmz, 2023-03-17:
Oh, I couldn't agree more @LeStarch ... I've been trying to port the project to RTEMS for 2 months and I still can't get it to work. Unless I'm mistaken, do we agree that there is no tutorial for porting to another OS?

I will wait until this issue is resolved to try porting again...

LeStarch, 2024-11-25:
Spun the last issue into the ROSES set!