nasa/fprime #1704: Consider simplifying the file names in assertion failure reports

bocchino, 2022-10-04:
We should consider simplifying our approach to using numeric file IDs vs. string file names when reporting assertion failures. The current approach uses conditional compilation to define the assertion reporting interface in one of two different ways. This approach causes ifdefs to be scattered all over the uses of the assertion interface. These ifdefs are hard to understand and maintain. See, e.g., https://github.com/nasa/fprime/pull/1703.

Instead, can we have the reporting mechanism pass a file ID and a pointer to the file name string, and let the client decide what to do (e.g., send file string or file ID to the ground)? If we did this, then instead of NO_ASSERT, FILE_ID_ASSERT, and FILENAME_ASSERT, we could just have a switch for assert on or off. If we wanted to have particular components (e.g., AssertFatalAdapter) with varying behavior (send down a string or an ID), we could add a config switch to just that component.

Rationale:

1. Omitting the string pointer argument may not be worth the maintenance cost, even if it’s not used. 

2. Simplifying the configuration switch seems good.

3. Even for applications that store or downlink the file ID, it is often useful to have the actual file name available for debugging. 

If there's a concern about using the __FILE__ string in the memory-constrained configuration, then we could keep the pointer in the interface but make it point to a dummy string in this configuration. I remember we made a similar simplification with object names. However, the system cost of passing a pointer to __FILE__ should not be high, even in a very constrained environment.

SterlingPeet, 2022-11-29:
On the smallest baremetal systems, it is possible for the length of a table of filenames to exceed the available program memory space.  While this _can_ be mitigated by truncating long paths, some of our systems are still too small for this.  LFPS and GT-1 both had a program memory of 128 kb for the entire deployment.

timcanham, 2022-11-30:
The motivation behind using `ASSERT_FILE_ID` instead was to have a config that compiles out any file name strings, which can take up a lot of room. @SterlingPeet's use case is exactly what it was intended for, so I wouldn't want to eliminate an approach with an integer-only version. Having a run-time decision would still leave the strings lingering in the build image.

bocchino, 2022-11-30:
These are good points. Thanks!

closed 2022-11-30 by bocchino