nasa/fprime #1705: Turning off `FW_ARRAY_TO_STRING` causes compile errors

timcanham, 2022-10-04:
| | |
|:---|:---|
|**_F´ Version_**| 3.1.1 |
|**_Affected Component_**| All  |
---
## Problem Description

Setting the `FW_ARRAY_TO_STRING` to 0 to eliminate the string function causes a build failure like:

```
[ 98%] Building CXX object CMakeFiles/Ref.dir/Top/Main.cpp.o
In file included from /home/tcanham/source/fprime/Ref/build-fprime-automatic-native/Ref/SignalGen/SignalGenComponentAc.hpp:37,
                 from /home/tcanham/source/fprime/Ref/SignalGen/SignalGen.hpp:20,
                 from /home/tcanham/source/fprime/Ref/build-fprime-automatic-native/Ref/Top/RefTopologyAc.hpp:15,
                 from /home/tcanham/source/fprime/Ref/Top/Main.cpp:6:
/home/tcanham/source/fprime/Ref/build-fprime-automatic-native/Ref/SignalGen/SignalPairSetArrayAc.hpp:132:27: error: "BUILD_UT" is not defined, evaluates to 0 [-Werror=undef]
  132 | #if FW_ARRAY_TO_STRING || BUILD_UT
      |                           ^~~~~~~~
In file included from /home/tcanham/source/fprime/Ref/build-fprime-automatic-native/Ref/SignalGen/SignalGenComponentAc.hpp:38,
                 from /home/tcanham/source/fprime/Ref/SignalGen/SignalGen.hpp:20,
                 from /home/tcanham/source/fprime/Ref/build-fprime-automatic-native/Ref/Top/RefTopologyAc.hpp:15,
                 from /home/tcanham/source/fprime/Ref/Top/Main.cpp:6:
/home/tcanham/source/fprime/Ref/build-fprime-automatic-native/Ref/SignalGen/SignalSetArrayAc.hpp:131:27: error: "BUILD_UT" is not defined, evaluates to 0 [-Werror=undef]
  131 | #if FW_ARRAY_TO_STRING || BUILD_UT
      |                           ^~~~~~~~
```

## How to Reproduce

1. Edit `config/FpConfig.hpp`
2. Change `FW_ARRAY_TO_STRING` to `0'
3. Compile

## Expected Behavior

The code should compile without errors or warnings.

timcanham, 2022-10-04:
Added the following lines to `FpConfig.hpp`:

```
#ifndef BUILD_UT
#define BUILD_UT 1
#endif
```

Then, these errors occur:

```
In file included from /home/tcanham/source/fprime/Fw/Types/Serializable.cpp:1:
/home/tcanham/source/fprime/Fw/Types/Serializable.hpp:35:25: error: ‘ostream’ in namespace ‘std’ does not name a type
   35 |             friend std::ostream& operator<<(std::ostream& os, const Serializable& val);
      |                         ^~~~~~~
/home/tcanham/source/fprime/Fw/Types/Serializable.hpp:9:1: note: ‘std::ostream’ is defined in header ‘<ostream>’; did you forget to ‘#include <ostream>’?
    8 | #include <FpConfig.hpp>
  +++ |+#include <ostream>
    9 | 
In file included from /home/tcanham/source/fprime/Fw/Types/Serializable.cpp:1:
/home/tcanham/source/fprime/Fw/Types/Serializable.hpp:138:25: error: ‘ostream’ in namespace ‘std’ does not name a type
  138 |             friend std::ostream& operator<<(std::ostream& os, const SerializeBufferBase& buff);
      |                         ^~~~~~~
/home/tcanham/source/fprime/Fw/Types/Serializable.hpp:138:20: note: ‘std::ostream’ is defined in header ‘<ostream>’; did you forget to ‘#include <ostream>’?
  138 |             friend std::ostream& operator<<(std::ostream& os, const SerializeBufferBase& buff);
      |                    ^~~

```

LeStarch, 2023-12-21:
There are several issues.  These need to be fixed in hand-code and FPP code. 

1. `#include` of string.hpp does not use || BUILD_UT
2. BUILD_UT should likely be (defined(BUILD_UT) && BUILD_UT) as BUILD_UT is not always set
3. `FW_SERIALIZABLE_TO_STRING_BUFFER_SIZE` is used when it is only defined by FW_SERIALIZABLE_TO_STRING=1

All three issues need to be fixed.

LeStarch, 2023-12-21:
Blocked by: https://github.com/fprime-community/fpp/issues/369

LeStarch, 2024-11-07:
Cause by [#2467](https://github.com/nasa/fprime/issues/2467)

matt392code, 2024-12-28:
**Proposed solution approach:**
First, standardize the header guards:
// Add this guard pattern consistently
#if defined(FW_ARRAY_TO_STRING) && FW_ARRAY_TO_STRING || (defined(BUILD_UT) && BUILD_UT)
    #include <string>
    // String-related functionality here
#endif

**Update the string buffer size configuration:**
// In FpConfig.hpp
#if defined(FW_SERIALIZABLE_TO_STRING) && FW_SERIALIZABLE_TO_STRING
    #define FW_SERIALIZABLE_TO_STRING_BUFFER_SIZE 256
#endif

// In code that uses the buffer size
#if defined(FW_SERIALIZABLE_TO_STRING) && FW_SERIALIZABLE_TO_STRING
    // Use FW_SERIALIZABLE_TO_STRING_BUFFER_SIZE
#endif

**Add proper BUILD_UT handling:**
#ifndef BUILD_UT
    #define BUILD_UT 0
#endif

**Fix the Serializable.hpp includes:**
#include <FpConfig.hpp>
#if defined(FW_ARRAY_TO_STRING) && FW_ARRAY_TO_STRING || (defined(BUILD_UT) && BUILD_UT)
    #include <ostream>
#endif

**The key principles here are:**
 - Always check if macros are defined before using them
 - Use consistent guards across all string-related functionality
 - Ensure configuration macros are only used when their related features are enabled
 - Add proper include guards for C++ stream functionality
This should allow the code to compile cleanly with FW_ARRAY_TO_STRING=0 while maintaining unit test functionality.

bocchino, 2025-01-08:
We have removed `FW_ARRAY_TO_STRING` and consolidated it with `FW_SERIALIZABLE_TO_STRING`.

closed 2025-01-08 by bocchino