nasa/fprime #1708: Remove vsnprintf from StringType.cpp - pulls in libraries for embedded

timcanham, 2022-10-04:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

A description of the problem with sufficient detail to understand the issue.

## How to Reproduce

1. 
2.
3.

## Expected Behavior

A description of the expected behavior.

LeStarch, 2022-11-30:
As a further note, we should probably review all `*printf*` usage

LeStarch, 2024-10-08:
Possible fixes:

1. `printf` -> `Fw::Logger::log()` and should be a 1-to-1 conversion. Don't do this in`Os/Log` implementations.
2. DEBUG_PRINT can be deleted where used
3. `snprintf` should be implemented as a wrapper in `Fw/Types/StringUtils.cpp` so the use is limited within that one `.cpp`

matt392code, 2024-12-28:
Recommendations for addressing each usage pattern:
**Debug Printing (DEBUG_PRINT)**
Current usage: Found in Linux driver implementations
Recommendation: Remove all DEBUG_PRINT macros as suggested, as they are commented out and not being used actively
Files to update:
LinuxSpiDriverComponentImpl.cpp
LinuxGpioDriverComponentImpl.cpp
LinuxUartDriver.cpp

String Formatting Operations (snprintf)
Most common usage (57 instances)
Recommendation: Create a centralized string formatting utility in Fw/Types/StringUtils.cpp:
namespace Fw {
    namespace StringUtils {
        // Safe string formatting with size checking
        NATIVE_INT_TYPE formatString(char* dest, NATIVE_INT_TYPE size, const char* format, ...);
        
        // Specialized formatters for common operations
        NATIVE_INT_TYPE formatComponentName(char* dest, NATIVE_INT_TYPE size, const char* name);
        NATIVE_INT_TYPE formatPortName(char* dest, NATIVE_INT_TYPE size, const char* name);
    }
}

**Error/Assertion Messages (fprintf/printf)**
Current usage: Primarily in Assert.cpp for error reporting
Recommendation: Replace with Fw::Logger::log() as suggested
Key files to update:
Fw/Types/Assert.cpp
Svc/AssertFatalAdapter/AssertFatalAdapterComponentImpl.cpp

**Logging Operations**
Replace printf-based logging with Fw::Logger::log()
Exception: Keep Os/Log implementations as is
Files needing updates:
Svc/PassiveConsoleTextLogger
Os/LogPrintf.cpp (keep as is - implementation file)

**vsnprintf Usage**
Single instance in StringType.cpp
Recommendation: Replace with custom implementation in StringUtils that doesn't pull in stdio:
// In Fw/Types/StringUtils.cpp
NATIVE_INT_TYPE Fw::StringUtils::formatStringV(char* dest, NATIVE_INT_TYPE size, const char* format, va_list args) {
    // Custom implementation without stdio dependency
    // Use basic string parsing and formatting
}

**Priority Implementation Steps:**
Create StringUtils Implementation
// Fw/Types/StringUtils.cpp
namespace Fw {
    namespace StringUtils {
        NATIVE_INT_TYPE formatString(char* dest, NATIVE_INT_TYPE size, const char* format, ...) {
            va_list args;
            va_start(args, format);
            NATIVE_INT_TYPE ret = formatStringV(dest, size, format, args);
            va_end(args);
            return ret;
        }
    }
}

**Update String Formatting Calls:**
Replace direct snprintf calls with Fw::StringUtils::formatString
Example conversion:
// Before
snprintf(buffer, size, "Component %s", name);

// After
Fw::StringUtils::formatString(buffer, size, "Component %s", name);

**Update Logging:**
// Before
printf("Hit Ctrl-C to quit\n");

// After
Fw::Logger::log(Fw::LogSeverity::INFO, "Hit Ctrl-C to quit");

**Benefits of This Approach:**
Centralized string formatting control
Reduced code size through elimination of printf library
Consistent logging interface
Better control over embedded system resources
Simplified future maintenance

**Testing Recommendations:**
Create unit tests for new StringUtils functions
Verify format string handling matches snprintf behavior
Test with various string lengths and formats
Verify error handling for buffer overflows

commit d13c7bf5 (2025-02-12): Fixes #1708 by Completing the work In `Fw` (#3173)

* Making string formatting into a fprime implementation choice

* Removing printf from Fw and cleaning-up toString methods

* Fixes to get UTs to compile

* Reverting test change to assert

* Removing error console and restoring assert functionality

* Create empty library to avoid 3.16 limitations on interface libs

* Turning off unused check on intentionally unused variable

* sp

* Fixing recursive calls and other CI problems

* Fixing compile issues
closed 2025-02-12 by LeStarch