nasa/fprime #1712: Raspberry PI (a $135 embedded Linux computer).

dlackner, 2022-10-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Due to the rising cost of inflation and global chip shortage, the Raspberry PI is no longer a $35 embedded Linux computer! I suggest updating the documentation in the README.md. =~$135 is a bit more accurate. 

## How to Reproduce

1. 
2.
3.

## Expected Behavior

A description of the expected behavior.

SterlingPeet, 2022-10-08:
Perhaps a slightly better approach is to support a handful of “low cost” embedded ARM Linux platforms instead.  Very minor tweaks (device strings) and a publicly available toolchain file put the same software on the BeagleBone Black.

LeStarch, 2022-10-10:
Not sure what the concern is.  Is it to switch to potential other chips, or to update our cost metric?

If the concern is to allow for other chips we could easily update our documentation. If the concern is our quoted price, we derive it from the official RPI website. Currently the official RPI website still quotes "From $35".

Prices higher than that are due to scalping and highly variable day-to-day.

LeStarch, 2022-11-30:
Closing this issue as no tangible work was agreed to.

closed 2022-11-30 by LeStarch