nasa/fprime #1713: Data Products

LeStarch, 2022-10-10:
### DpCatalog

`DpCatalog` will do the following:

1\) Upon initialization, read the DP directory and build a catalog of products
2\) When commanded, start downloading based on priority and other TBD criterion (date? size?)
3\) Manage deletion of data products (commands or file list)
4\) Generate DP listings for download (file? self-defined DP?)

This was derived from:

_Originally posted by @timcanham in https://github.com/nasa/fprime/discussions/1154_

matt392code, 2025-01-03:
[dp-catalog-impl.py.txt](https://github.com/user-attachments/files/18304572/dp-catalog-impl.py.txt)
Proposed implementation:
1. Core Data Structures:
   - DpMetadata: Serializable structure storing essential file information
   - Internal catalog dictionary mapping filenames to metadata
   - Priority-based download queue
2. Main Component Operations:
   - Directory scanning and catalog building
   - Prioritized download management
   - File deletion handling
   - Status reporting and listing generation
3. F Prime Integration:
   - Uses F Prime's serializable types for command/telemetry interfaces
   - Implements command handlers for ground system control
   - Provides telemetry points for monitoring
4. Design Considerations:
   - Thread-safe operations for concurrent access
   - Efficient directory scanning using periodic updates
   - Flexible criteria-based download prioritization
   - Multiple output format support for listings
The implementation leaves certain aspects configurable:
- Priority calculation algorithms
- Directory scanning frequency
- Download concurrency limits
- Storage management policies

LeStarch, 2025-01-15:
I believe this was delivered.

closed 2025-01-15 by LeStarch