nasa/fprime #1724: Clickable UT Assert Messages

LeStarch, 2022-10-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Currently F´ unittest assert messages look like the following:

```
File: <file>
Line: ###
```

However, IDEs typically make messages automatically clickable if they are of the form:
```
<file>:###
```
GoogleTest asserts follow this form, and we should switch.

SMorettini, 2022-11-10:
If someone didn't pick it up yet, I can do it :)

LeStarch, 2022-11-11:
@SMorettini go for it!

SMorettini, 2022-12-30:
I create the pull request. I update the messages for the asserts in the UT. Should I change also the messages in the normal asserts?

SMorettini, 2023-01-13:
@LeStarch Should I apply the same change to the other ASSERTS? If yes should I do another issue or link to this one?

LeStarch, 2023-01-16:
@SMorettini I opened up #1847 to track this. However, let's not implement just yet.  I need to think through the implications.

SMorettini, 2023-01-17:
Yes, I was referring to FW_ASSERTS. Since you create specific issues for that, I think you can proceed with closing this issue(I can't because you made it).

PR #1829 (2023-01-13): Clickable ut assert messages #1724
| | |
|:---|:---|
|**_Affected Component_**| Autocoder  |
|**_Affected Architectures(s)_**| Unit tests |
|**_Related Issue(s)_**| #1724  |

---
## Change Description

Close #1724
closed 2023-01-17 by LeStarch