nasa/fprime #1728: Fail to build UTs when disabling FW_ENABLE_TEXT_LOGGING

kevin-f-ortega, 2022-10-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Get this error 
```
build-fprime-automatic-native-ut/F-Prime/Autocoders/Python/templates/TesterBase.cpp:1081:11: error: ���class ExampleComponents::ExampleTesterBase��� has no member named ���textLogHistory���

 1081 |     this->textLogHistory->clear();

      |           ^~~~~~~~~~~~~~
```
## How to Reproduce

1. Disable FW_ENABLE_TEXT_LOGGING in config/FpConfig.hpp
2. Run fprime-util check --all -j 4

## Expected Behavior

A description of the expected behavior.

LeStarch, 2023-12-21:
This config value is required to build most UTs.  Thus, we either need to remove this configuration setting, ensure that it is always on for UTs, or fix UTs

LeStarch, 2023-12-21:
Needed UT update example:
```
+#if FW_ENABLE_TEXT_LOGGING == 1
         // LogText
         this->component.set_LogText_OutputPort(0, this->get_from_LogText(0));
+#endif
```

matt392code, 2024-12-28:
Based on LeStarch's comment, we need to wrap text logging related code in the unit tests with the appropriate conditional compilation guards. This pattern should be applied consistently:
```
#if FW_ENABLE_TEXT_LOGGING == 1
    // Text logging related code here
    this->textLogHistory->clear();
#endif
```
**Implementation Steps:**
Find all instances in TesterBase where text logging is accessed
Add the conditional compilation guards around:
Member declarations
Method implementations
Port connections (as shown in LeStarch's example)
Any clear() or setup calls

**Example Fix for the Error:**
```
// In the TesterBase class declaration
class ExampleTesterBase {
#if FW_ENABLE_TEXT_LOGGING == 1
    History<TextLogEntry>* textLogHistory;
#endif
    // ... other members
};
// In the implementation
void ExampleTesterBase::clearHistory() {
#if FW_ENABLE_TEXT_LOGGING == 1
    this->textLogHistory->clear();
#endif
    // ... clear other histories
}
```
Additional Considerations:
Need to review if there are any test assertions or verifications that depend on text logging
Those would also need to be conditionally compiled
May want to add a warning or note in the test documentation about reduced test coverage when text logging is disabled

bocchino, 2025-01-08:
I vote for requiring that text logging is on for UTs. See #2467. The alternative causes a proliferation of ifdefs and is challenging to maintain. Also the FSW coding rules tell us not to use these ifdefs. We use them when we have to, but we should avoid them if possible.

bocchino, 2025-01-08:
Actually, it looks like this issue pertains to the old Python autocoder and has been fixed in the FPP autocoder. In the code generated by `fpp-to-cpp`, the guard is there. Also the FPP unit tests run the compilation with the guard on and off. So I think we can close this one as fixed by FPP.

closed 2025-01-08 by bocchino