nasa/fprime #1732: CMake errors encountered while attempting to build Ref with MathComponent

G5STER, 2022-10-21:
| | |
|:---|:---|
|**_F´ Version_**| 3.1.1|
|**_Affected Component_**|   Ref and MathComponent|
---
## Problem Description

Encountered following CMake errors when attempting to build Ref with MathComponent
```
In file included from /home/kasm-user/fprime/Ref/build-fprime-automatic-native/Ref/Top/RefTopologyAc.cpp:7:
/home/kasm-user/fprime/Ref/build-fprime-automatic-native/Ref/Top/RefTopologyAc.hpp:271:15: error: ‘ByteStreamDriverModel’ in namespace ‘Drv’ does not name a type
  271 |   extern Drv::ByteStreamDriverModel comm;
      |               ^~~~~~~~~~~~~~~~~~~~~
/home/kasm-user/fprime/Ref/build-fprime-automatic-native/Ref/Top/RefTopologyAc.hpp:301:15: error: ‘Time’ in namespace ‘Svc’ does not name a type
  301 |   extern Svc::Time linuxTime;
      |               ^~~~
make[3]: *** [Ref/Top/CMakeFiles/Ref_Top.dir/build.make:302: Ref/Top/CMakeFiles/Ref_Top.dir/RefTopologyAc.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:8390: Ref/Top/CMakeFiles/Ref_Top.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2045: CMakeFiles/Ref.dir/rule] Error 2
make: *** [Makefile:195: Ref] Error 2
[ERROR] CMake erred with return code 2
```

## How to Reproduce

1. Checkout v3.1.1
2. Copy MathComponent contents to Ref
3. Generate stubs using `fprime-util generate` in Ref directory
4. Build using `fprime-util build`

See https://github.com/ODU-CGA-CubeSat/fprime/commit/87b8851a398a02540df996491955d80ed424b354

## Expected Behavior

Compile without errors

LeStarch, 2022-10-28:
Were there any changes to the `Ref/Top` files?

thomas-bc, 2023-11-29:
Outdated

closed 2023-11-29 by thomas-bc