nasa/fprime #1738: Build error: "Input object has no document: lxml.etree._XSLTResultTree"

crmauceri, 2022-10-26:
| | |
|:---|:---|
|**_F´ Version_**|  master |
|**_Affected Component_**|   |
---
## Problem Description

I'm following the installation instructions, but when I try the build step (`fprime-util build --jobs 10`), I get the following error. 

```
[ 53%] Generating AssertFatalAdapterComponentAc.hpp, AssertFatalAdapterComponentAc.cpp
Input object has no document: lxml.etree._XSLTResultTree
Traceback (most recent call last):
  File "/Users/cmauceri/My Drive/Workspace/FPrime/fprime/Autocoders/Python/bin/codegen.py", line 1334, in <module>
    main()
  File "/Users/cmauceri/My Drive/Workspace/FPrime/fprime/Autocoders/Python/bin/codegen.py", line 1259, in main
    the_parsed_component_xml = XmlComponentParser.XmlComponentParser(
  File "/Users/cmauceri/My Drive/Workspace/FPrime/fprime/Autocoders/Python/src/fprime_ac/parsers/XmlComponentParser.py", line 113, in __init__
    self.validate_xml(xml_file, element_tree, "schematron", "active_comp")
  File "/Users/cmauceri/My Drive/Workspace/FPrime/fprime/Autocoders/Python/src/fprime_ac/parsers/XmlComponentParser.py", line 1218, in validate_xml
    validator_compiled = isoschematron.Schematron(validator_parsed)
  File "/Users/cmauceri/My Drive/Workspace/FPrime/fprime-venv/lib/python3.9/site-packages/lxml/isoschematron/__init__.py", line 275, in __init__
    schematron = self._expand(schematron, **expand_params)
  File "src/lxml/xslt.pxi", line 515, in lxml.etree.XSLT.__call__
  File "src/lxml/apihelpers.pxi", line 43, in lxml.etree._documentOrRaise
ValueError: Input object has no document: lxml.etree._XSLTResultTree
make[3]: *** [F-Prime/Svc/AssertFatalAdapter/AssertFatalAdapterComponentAc.hpp] Error 255
make[2]: *** [F-Prime/Svc/AssertFatalAdapter/CMakeFiles/Svc_AssertFatalAdapter.dir/all] Error 2
make[1]: *** [CMakeFiles/Ref.dir/rule] Error 2
make: *** [Ref] Error 2
[ERROR] CMake erred with return code 2
```
## Details

I'm on a Mac with an M1 chip running Monteray. 

I used `fprime-util build --jobs 10` instead of `fprime-util build --jobs "$(nproc || printf '%s\n' 1)"` because nproc is not available on mac. 

## How to Reproduce

1. Follow [Installation Instructions](https://github.com/nasa/fprime/blob/master/docs/INSTALL.md) 
2. Error occurs when I run `fprime-util build --jobs "$(nproc || printf '%s\n' 1)"`

## Expected Behavior

I expect the build to execute successfully.

crmauceri, 2022-10-28:
Just confirmed that this error also occurs without the jobs flag.

LeStarch, 2022-10-28:
What is the version of Python you have?  This is an issue in our python tooling although I am unsure yet why it occurs.

crmauceri, 2022-10-28:
I've got Python 3.9

LeStarch, 2022-10-28:
I think I have a solution.  It appears that updating any of the setuptools/pip packages on Mac M1 breaks the install.

Start with a clean terminal, and rerun the installation (new virtual environment and everything).  **Emit** the following command:

```
pip install -U setuptools setuptools_scm wheel pip
```
This is the command that breaks M1 installations...but ironically is absolutely required for other systems to function at all.

You may wish to add `--no-cache` to any `pip install` commands to force it to completely start-over.

crmauceri, 2022-10-31:
@LeStarch Can you share what version of setuptools setuptools_scm wheel and pip work for you? 

In my new environment, I've got 
```
Name: pip
Version: 22.0.4

Name: setuptools
Version: 58.1.0

WARNING: Package(s) not found: setuptools_scm

WARNING: Package(s) not found: wheel
```

But I'm still getting the same error. I'm guessing the default packages for setuptools and pip are newer on my system than on yours.

crmauceri, 2022-10-31:
Actually, let me provide my full package list. What lxml version do you have? Since it's mentioned in the error, differing `lxml.etree._XSLTResultTree` versions might be the problem.

```
Package            Version
------------------ -----------
aniso8601          9.0.1
argcomplete        1.12.3
arrow              1.2.2
attrs              21.4.0
binaryornot        0.4.4
Brotli             1.0.9
certifi            2022.6.15
chardet            4.0.0
charset-normalizer 2.0.12
Cheetah3           3.2.6.post1
click              7.1.2
cookiecutter       1.7.3
Flask              1.1.4
Flask-Compress     1.12
Flask-RESTful      0.3.9
fprime-fpp         1.0.2
fprime-gds         3.1.3
fprime-tools       3.1.1
gcovr              5.1
idna               3.3
importlib-metadata 4.11.4
iniconfig          1.1.1
itsdangerous       1.1.0
Jinja2             2.11.3
jinja2-time        0.2.0
lxml               4.9.0
Markdown           3.3.7
MarkupSafe         2.0.1
packaging          21.3
pexpect            4.8.0
pip                22.0.4
pluggy             1.0.0
poyo               0.5.0
ptyprocess         0.7.0
py                 1.11.0
Pygments           2.12.0
pyparsing          3.0.9
pytest             6.2.5
python-dateutil    2.8.2
python-slugify     6.1.2
pytz               2022.1
pyzmq              22.3.0
requests           2.28.0
setuptools         58.1.0
six                1.16.0
text-unidecode     1.3
toml               0.10.2
urllib3            1.26.9
Werkzeug           1.0.1
zipp               3.8.0
```

LeStarch, 2022-10-31:
What command did you use to print that out?

crmauceri, 2022-10-31:
pip list

LeStarch, 2022-10-31:
Here is what I have:

**OS:** macOS Monterrey (12.6)
**Python:** Python 3.9.6 (OS supplied)

## In NewTerminal
```
python3 -m venv fprime-temp
. fprime-temp/bin/activate
pip install --no-cache -r ../code/fprime/requirements.txt

cd fprime/Ref
fprime-util generate
fprime-util build
```
> I am using `--no-cache` to ensure I am getting a fresh copy of anything downloaded.

## PIP List
```
Package            Version
------------------ -----------
aniso8601          9.0.1
argcomplete        1.12.3
arrow              1.2.2
attrs              21.4.0
binaryornot        0.4.4
Brotli             1.0.9
certifi            2022.6.15
chardet            4.0.0
charset-normalizer 2.0.12
Cheetah3           3.2.6.post1
click              7.1.2
cookiecutter       1.7.3
Flask              1.1.4
Flask-Compress     1.12
Flask-RESTful      0.3.9
fprime-fpp         1.0.2
fprime-gds         3.1.3
fprime-tools       3.1.1
gcovr              5.1
idna               3.3
importlib-metadata 4.11.4
iniconfig          1.1.1
itsdangerous       1.1.0
Jinja2             2.11.3
jinja2-time        0.2.0
lxml               4.9.0
Markdown           3.3.7
MarkupSafe         2.0.1
packaging          21.3
pexpect            4.8.0
pip                21.2.4
pluggy             1.0.0
poyo               0.5.0
ptyprocess         0.7.0
py                 1.11.0
Pygments           2.12.0
pyparsing          3.0.9
pytest             6.2.5
python-dateutil    2.8.2
python-slugify     6.1.2
pytz               2022.1
pyzmq              22.3.0
requests           2.28.0
setuptools         58.0.4
six                1.16.0
text-unidecode     1.3
toml               0.10.2
urllib3            1.26.9
Werkzeug           1.0.1
zipp               3.8.0
```

crmauceri, 2022-10-31:
I made a new environment exactly matching the python package versions, including pip and setuptools, and I'm still seeing this problem.

LeStarch, 2022-10-31:
What is the subversion of Python?  You said python 3.9, but is it 3.9.6?

crmauceri, 2022-10-31:
Aha, that might be it. I have Python 3.9.12. I'll try again with 3.9.6

LeStarch, 2022-10-31:
Is `3.9.12` provided via macO or something you downloaded?  It could be that the macOS provided python is better equipped for M1 macs....

crmauceri, 2022-10-31:
Python 3.9.6 works! 

For future reference, the full version information of both versions is

Python 3.9.12 (main, Apr  5 2022, 01:52:34) 
[Clang 12.0.0 ] :: Anaconda, Inc. on darwin

Python 3.9.6 (default, Sep 13 2022, 22:03:16) 
[Clang 14.0.0 (clang-1400.0.29.102)] on darwin

closed 2022-10-31 by crmauceri