nasa/fprime #1751: Script to generate parameter commands

timcanham, 2022-11-01:
| | |
|:---|:---|
|**_F´ Version_**|3.x |
|**_Affected Component_**| All |
---
## Feature Description

Add script to generate parameter setting sequence. The script can use default value where specified, and put error-causing argument in one where they are not to force the user to fill them in. The script can use the XML dictionary.

## Rationale

An automated way to intialize parameters without requiring the user to go component by component and manually build a list.

matt392code, 2024-12-28:
[fprime-param-script.py.txt](https://github.com/user-attachments/files/18264589/fprime-param-script.py.txt)
Python script that addresses the feature request. Here's how it works:
1. The script parses F Prime XML dictionary files to extract parameter information including:
   - Component and parameter names
   - Parameter types
   - Default values
   - Valid ranges
   - Descriptions
2. For parameters without default values, it generates error-causing values specific to the parameter type to force user attention.

3. Key features:
   - Generates complete parameter setting sequences using `fprime-cli parameter-set` commands
   - Includes helpful warnings and documentation about parameters needing user input
   - Supports optional generation of commands for all parameters via `--force-all` flag
   - Preserves parameter constraints and descriptions in the output
   - Handles different parameter types appropriately
To use the script:
```bash
python3 generate_params.py <path_to_xml_dict> <output_file> [--force-all]
```
The output file will contain:
- Warning section listing parameters requiring user attention
- Parameter ranges and descriptions where available
- Generated command sequence

The script helps automate parameter initialization while ensuring users are aware of which parameters need their attention.

LeStarch, 2025-02-04:
@zimri-leisher this was a future feature of your param script.

zimri-leisher, 2025-08-20:
merged in nasa/fprime-gds#187