nasa/fprime #1752: Fpp constants available at compile-time

mohamed-ghita, 2022-11-02:
| | |
|:---|:---|
|**_3.x**| |
|**All**|  |
---
## Feature Description

Make Fpp constants available at compile-time by auto-coding them as `constexpr` variables.

## Rationale

Fpp constants are auto coded as either `enum` member for integral constants or `extern const` for floating-point. 
`extern const` variables are not available at compile time. For example, `static_assert(FPP_F32_CONSTANT > 0);` will fail to compile because value of `FPP_F32_CONSTANT` is available only during linkage.

Also, it's cleaner to use `constexpr` for both integrals and floating-points instead of relying on enums. 

Current FppConstantAC.hpp:
```
namespace ModuleNamespace {
      enum FppConstant_CONSTANT_I32{
        CONSTANT_I32 = 5
      };
      extern const F32 CONSTANT_F32; // value assigned in .cpp file
}
```

with constexpr:
```
namespace ModuleNamespace {
      constexpr I32 CONSTANT_I32{5};
      constexpr F32 CONSTANT_F32{5.5F};
}
```

codeflight1, 2022-11-02:
I believe this hasn't been done because `constexpr` is a C++11 feature.

ThibFrgsGmz, 2022-11-02:
According to the [coding style](https://github.com/nasa/fprime/blob/master/docs/UsersGuide/dev/code-style.md#c), C++ 11 is the official project standard, but some features are avoided - perhaps `constexpr` is one of them.

mohamed-ghita, 2022-11-03:
According to the [coding guidlines](https://github.com/nasa/fprime/blob/master/docs/UsersGuide/dev/code-style.md#embedded-c-usage), developers are encouraged to used `std::numeric_limits` which defines `constexpr` methods such as `static constexpr T std::numeric_limits<T>::max(); since C++11` [cpp refrence](https://en.cppreference.com/w/cpp/types/numeric_limits/max).

bocchino, 2023-06-29:
I agree it would be better to use constexpr. The FPP constant code gen was written back when we were still using C++98.

LeStarch, 2024-03-28:
@bocchino is this one on your radar?

bocchino, 2024-03-29:
It's an open issue in FPP (#269). See above.

LeStarch, 2024-03-29:
I am going to close this one since it is tracked directly in FPP.

closed 2024-03-29 by LeStarch