nasa/fprime #1753: Add Parameter dump feature to autocoder.

timcanham, 2022-11-02:
| | |
|:---|:---|
|**_F´ Version_**| 3.X|
|**_Affected Component_**| All |
---
## Feature Description

Add events for all parameters that dump the parameter value.

## Rationale

It allows inspection of actual values set, which currently requires user-written code.

SterlingPeet, 2022-12-04:
Please make this feature optional, but enabled by default.  Our very small systems leverage Parameters, and it would be preferable to disable this feature for code space savings.

timcanham, 2023-03-02:
A follow up to @SterlingPeet's idea: if we don't specify the dump command opcode, don't generate the code.

LeStarch, 2023-03-02:
This may make reuse of components that define that dump command harder as you'd need to clone the model and strip out that one tag.  A global "no parameter dumping" switch may be better for these systems.

matt392code, 2024-12-28:
[param-dump-implementation.py.txt](https://github.com/user-attachments/files/18264603/param-dump-implementation.py.txt)
The key aspects of this implementation:
**1.  Configuration Approach:**
Added a global ParamDumpConfig with ENABLED/DISABLED modes
Default enabled but can be disabled for resource-constrained systems
Configuration through FPP (F Prime Prime) component definition
**2.  Parameter Dump Features:**
Automatic event generation on parameter updates
Two command handlers:
a.  DUMP_ALL_PARAMS: Dumps all parameter values
b.  DUMP_PARAM: Dumps a specific parameter value
Event logging with parameter ID and string representation
**3.  Code Generation:**
Conditional code generation based on PARAM_DUMP_MODE
Generates parameter update hooks for event emission
Generates command handlers for dump commands
Proper error handling for invalid parameter IDs
**4.  Resource Optimization:**
When DISABLED, no code is generated for dump functionality
Minimal overhead when enabled (just event logging)
Reusable component structure maintained
**5.  Extensibility:**
Easy to add new parameters without modifying dump logic
Structure allows for future enhancements (e.g., filtering, rate limiting)
**This implementation addresses the concerns raised in the discussion:**
Supports SterlingPeet's request for optional generation
Implements timcanham's suggestion for conditional code generation
Maintains LeStarch's component reusability through global configuration