nasa/fprime #1766: Unit test implementation fails when FW_SERIALIZABLE_TO_STRING is set

Austen-G, 2022-11-09:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description
When FW_SERIALIZABLE_TO_STRING is set, the PolyType class throws an error that no declaration matches: `void Fw::PolyType::toString(Fw::StringBase&) const`

After a bit of digging it appears to be because it expects a fixed width integer, but NATIVE_INT_TYPE is not defined as fixed width.

## How to Reproduce

1. Set FW_SERIALIZABLE_TO_STRING
2. Generate unit test build cache
3. Run fprime-util impl --ut from any component directory with a unit-test section in its CMakeLists.txt

## Expected Behavior

Tester.cpp and Tester.hpp files should be successfully generated.

LeStarch, 2023-12-21:
There are several issues.  These need to be fixed in hand-code and FPP code. 

1. `#include` of string.hpp does not use || BUILD_UT
2. BUILD_UT should likely be (defined(BUILD_UT) && BUILD_UT) as BUILD_UT is not always set
3. `FW_SERIALIZABLE_TO_STRING_BUFFER_SIZE` is used when it is only defined by FW_SERIALIZABLE_TO_STRING=1

All three issues need to be fixed.

Related to #1705, #1727

LeStarch, 2023-12-21:
Blocked by: https://github.com/fprime-community/fpp/issues/369

matt392code, 2024-12-28:
Proposed Solutions:
**1. For the string.hpp include issue:**
```cpp
#if defined(FW_SERIALIZABLE_TO_STRING) || defined(BUILD_UT)
#include <Fw/Types/StringType.hpp>
#endif
```
**2. For the BUILD_UT macro usage:**
```cpp
#if (defined(BUILD_UT) && BUILD_UT) || defined(FW_SERIALIZABLE_TO_STRING)
// String conversion code
#endif
```
**3. For the buffer size issue, we should add a fallback definition:**
```cpp
#if !defined(FW_SERIALIZABLE_TO_STRING) || !defined(FW_SERIALIZABLE_TO_STRING_BUFFER_SIZE)
#define FW_SERIALIZABLE_TO_STRING_BUFFER_SIZE 256 // Default size for unit tests
#endif
```
**The fixes need to be applied in:**
1. The PolyType class implementation
2. Any FPP-generated code that uses these features
3. Related unit test files
**Implementation Steps:**
1. Update the conditional includes in affected header files
2. Modify the macro checks to properly handle BUILD_UT cases
3. Add fallback buffer size definition for unit test scenarios
4. Update FPP code generation templates if needed to handle these conditions
This should allow unit tests to compile and run properly regardless of the FW_SERIALIZABLE_TO_STRING setting while maintaining the original functionality for non-test builds.

bocchino, 2025-01-08:
Superseded by https://github.com/nasa/fprime/issues/2467.