nasa/fprime #1767: LinuxSpiDriver bug

gcgandhi, 2022-11-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

The `LinuxSpiDriver` had a bug introduced here:  https://github.com/nasa/fprime/commit/ec191ae52839ee9c4ca9ba2b2553a3ac662f77b1#diff-f26408567a783542946bca72183d884a11c52193b4f3e32afc579ece19c12a78

Issue is that the pointers to the buffers should be `unsigned long`, which is typically 32 bits, and it was casted to a `U64`.  Casting to a `unsigned long` is working for me on our platform.

## How to Reproduce

1. Run the driver and see a bad status returned from SPI read/writes
2.
3.

## Expected Behavior

A description of the expected behavior.

LeStarch, 2022-11-29:
This seems to mean that the SPI struct object uses platform-dependent int types.  We should explicitly cast to the header defined type.  If that is `unsigned long` so be it.

LeStarch, 2022-11-29:
@gcgandhi if you have a fix, do you want to send it back?

gcgandhi, 2022-11-29:
@LeStarch Fix was trivial (if you have access):  https://cadregitlab.jpl.nasa.gov/cadre/cadre-fsw/-/commit/a94d9d6e28571a6b1ee88f5d362959587729f5da

Just changed to:
```
 tr.tx_buf = reinterpret_cast<unsigned long>(writeBuffer.getData());
 tr.rx_buf = reinterpret_cast<unsigned long>(readBuffer.getData());
```

LeStarch, 2022-11-29:
@gcgandhi thanks!

LeStarch, 2022-11-30:
@gcgandhi in all the headers I find online i see the definition:

```
	__u64		tx_buf;
	__u64		rx_buf;
```
Which makes it look like it is specifically 64 bit.  Can you send the definitions your platform has?

gcgandhi, 2022-11-30:
@LeStarch Yea when I digged deeper the `__u64` is a `unsigned long`, which is why I used that.

LeStarch, 2022-11-30:
@gcgandhi, I think this is fixed in devel.  It uses a proper reinterpret cast:

```
        tr.tx_buf = reinterpret_cast<U64>(writeBuffer.getData());
        tr.rx_buf = reinterpret_cast<U64>(readBuffer.getData());
```

gcgandhi, 2022-11-30:
@LeStarch No that is not correct.  A `U64` is 8 bytes and our platform a `unsigned long` is 4 bytes.  Typically a `unsigned long` is 4 bytes and a `unsigned long long` is 8 bytes.

LeStarch, 2022-11-30:
@gcgandhi still, my question is what type your `spidev.h` header uses, as all examples I found use a U64.  If this is the case, then this issue can be closed.  If the type is different on your platform, then we need to fix it.

gcgandhi, 2022-11-30:
@LeStarch our platform uses `unsigned long` for `__u64`, but based on the comments I see above the structure sounds like it can be either 32 bit or 64 bits.  Perhaps cast it to `__u64`?

LeStarch, 2022-11-30:
@gcgandhi casting to `__u64` seems reasonable as that is the "real" type.  I will do that.

closed 2023-01-17 by LeStarch