nasa/fprime #1789: `fprime-util build` fails to regenerate binary if only linker script changes

kbotteon, 2022-11-30:
| | |
|:---|:---|
|**_F´ Version_**| 3.1.1 |
|**_Affected Component_**| Linker |
---
## Problem Description

If only the linker script is updated, `fprime-util build` will not generate a new binary. A source file must be changed.

## How to Reproduce

1. Edit `cmake/ldscripts/lscript.ld`
2. Run `fprime-util build`
3. Observe that the binary is stale. For example, you can move a section like `.special_user_data` to a different memory region (e.g. from SRAM to Flash) using the linker script and then check its location with `mb-objdump -tC </path/to/binary>`

## Expected Behavior

Changes to the linker script are important and can fundamentally alter the binary. The binary should be regenerated when changes are made.

LeStarch, 2022-11-30:
How is that .ld script engaged?  Is it set in the toolchain somehow?  If so, we likely need to add a CMake command to the toolchain to detect changes to that file.

kbotteon, 2022-11-30:
In our case, we have the equivalent of https://github.com/nasa/fprime/blob/master/cmake/toolchain/raspberrypi.cmake for our toolchain. The script is specified with `set(CMAKE_EXE_LINKER_FLAGS ...)` command.

LeStarch, 2022-11-30:
Yes.  The CMake way to fix it is discussed here: https://stackoverflow.com/questions/32864689/cmake-how-to-add-dependency-on-linker-script-for-executable

In short, set a target property "LINK_DEPENDS" to a list of files.  You can do this in your deployment cmake:

```
register_deployment()
set_target_properties(${PROJECT_NAME} PROPERTIES LINK_DEPENDS ${LINKER_SCRIPT})
```

LeStarch, 2022-11-30:
I agree this is a sufficiently common problem that wrapping it in a fprime build system syntactic construct would be nice.  E.g. you'd specify in the toolchain or platform:

```
set(FPRIME_LINK_DEPENDS ...)
```
...and it would automatically fill this in to each target.  Until this is built, we can keep this issue open.

kbotteon, 2022-11-30:
Great, adding the additional CMake command resolve this issue.

I agree that a `LINK_DEPENDS` would be nice, or inferring that based on `CMAKE_EXE_LINKER_FLAGS`, because I also expect to encounter this issue when updating external static libraries, since those are also only referenced via the linker command given to CMake.

matt392code, 2024-12-28:
[fprime-linker-fix.txt](https://github.com/user-attachments/files/18264644/fprime-linker-fix.txt)
Looking at the raspberrypi.cmake toolchain file, it's currently set up as a basic ARM cross-compilation toolchain without linker script dependency tracking. The key fix would be to add proper dependency tracking using CMake's LINK_DEPENDS property.
**Implementation:**
This solution provides several key improvements:
- It adds explicit linker script dependency tracking using CMake's LINK_DEPENDS property
- It introduces a new FPRIME_LINK_DEPENDS cache variable to allow additional dependencies to be specified
- It automatically detects linker scripts specified via -T flags in CMAKE_EXE_LINKER_FLAGS
- It maintains backward compatibility with existing F Prime builds

The implementation provides two ways to handle linker script dependencies:
a.  Explicit: Adding files to FPRIME_LINK_DEPENDS
b.  Automatic: Detecting scripts specified in CMAKE_EXE_LINKER_FLAGS

This should resolve the issue where modifying the linker script doesn't trigger a rebuild. The binary will now be regenerated whenever the linker script changes, ensuring memory layouts are always correct.