nasa/fprime #1791: Svc::ComQueue fails to compile

kevin-f-ortega, 2022-12-02:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  `Svc::ComQueue`  |
---
## Problem Description

When compiling for bare metal, I get this error

```
~/myproject/fprime/Svc/ComQueue/ComQueue.cpp: In member function 'void Svc::ComQueue::configure(Svc::ComQueue::QueueConfigurationTable, NATIVE_UINT_TYPE, Fw::MemAllocator&)':
~/myproject/fprime/Svc/ComQueue/ComQueue.cpp:94:76: error: cannot bind non-const lvalue reference of type 'NATIVE_UINT_TYPE&' {aka 'short unsigned int&'} to an rvalue of type 'NATIVE_UINT_TYPE' {aka 'short unsigned int'}
   94 |     this->m_allocation = this->m_allocator->allocate(this->m_allocationId, totalAllocation, recoverable);
      |                                                                            ^~~~~~~~~~~~~~~
In file included from ~/myproject/fprime/Svc/ComQueue/ComQueue.hpp:14,
                 from ~/myproject/fprime/Svc/ComQueue/ComQueue.cpp:8:
~/myproject/fprime/Fw/Types/MemAllocator.hpp:57:39: note:   initializing argument 2 of 'virtual void* Fw::MemAllocator::allocate(NATIVE_UINT_TYPE, NATIVE_UINT_TYPE&, bool&)'
   57 |                     NATIVE_UINT_TYPE &size,
      |                     ~~~~~~~~~~~~~~~~~~^~~~
make[2]: *** [F-Prime/Svc/ComQueue/CMakeFiles/Svc_ComQueue.dir/build.make:132: F-Prime/Svc/ComQueue/CMakeFiles/Svc_ComQueue.dir/ComQueue.cpp.obj] Error 1
```

## How to Reproduce

1. Compile with a bare metal compiler
2.
3.

## Expected Behavior

I can compile for bare metal.

bocchino, 2022-12-08:
This should be fixed in #1803.

kevin-f-ortega, 2022-12-08:
I agree. Thank you @bocchino !

PR #1803 (2022-12-08): Revise Linux platform types
* Simplify the code; similar to #1800
* Add an explicit error for the "should not happen" case that `__SIZEOF_POINTER__` is not in { 8, 4, 2, 1 }. Before the code was falling through to 1 without checking that 1 was correct.
* Incidentally fix a regression in the test autocoder. See comment below.
* Incidentally fix #1791
closed 2022-12-08 by kevin-f-ortega