nasa/fprime #1801: fprime-util fpp-check interaction issue

LMPS97, 2022-12-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

When following the math component tutorial, section 6.2, running the command `fprime-util fpp-check -u unconnected.txt` causes an error:

``` bash
fpp-check
/Users/lsoares/fprime/Drv/BlockDriver/Tlm.fppi: 2.1
telemetry BD_Cycles: U32 id 0
^
error: module member expected
```

## How to Reproduce

1. Build master branch of fprime as described by the math component tutorial
2. `cd Ref/Top`
3. `fprime-util fpp-check -u unconnected.txt`

## Expected Behavior

The `unconnected.txt` should be populated with the unconnected ports of the project.

bocchino, 2022-12-07:
@LeStarch This looks like an issue with the way that `fprime-util` is collecting dependencies and submitting them to fpp-check. The FPP tool usage should look something like this:
```
% fpp-depend ../build-fprime-automatic-native/locs.fpp instances.fpp topology.fpp > depends.fpp
% fpp-check -u unconnected.txt instances.fpp topology.fpp `cat depends.fpp`
```
When I run these commands manually, they work as expected.

LeStarch, 2023-01-04:
`fprime-util` reads from the fpp-input-list cache file.  This file is not generated in the same way as the above checks.

thomas-bc, 2023-02-01:
@LMPS97 this issue should be resolved with the merging of #1872. Could you confirm on your end (with latest `devel`)?
Thank you!

bocchino, 2023-02-08:
Thanks, Thomas. So far I have done the following:

1. Pull down the latest devel.
2. Re-install F Prime
3. Wipe out and regenerate the automatic-native build cache in Ref
4. Run `fprime-util fpp-check -u unconnected.txt`

I get this error message:
```
[I] bocchino@MT-306179 ~/J/T/F/f/Ref (devel)> fprime-util fpp-check -u unconnected.txt
[ERROR] fpp-* commands require fprime v3.1.0+. Assumed older version because /Users/bocchino/JPL/Tools/FPrime/fprime/Ref/build-fprime-automatic-native/Ref/fpp-input-list not found
```
I think my F Prime version is 3.1.2a1.

bocchino, 2023-02-08:
I confirmed that `fprime-util fpp-check -u unconnected.txt` works as expected when run in `Ref/Top`. When I run it in `Ref`, I get the error message reported above. This message is confusing. I think the actual error is that there is no FPP model in `Ref`.

thomas-bc, 2023-02-08:
Yes the `fprime-util fpp-check` command needs to be ran from the topology directory. Is that behavior ok? I can add an option to point to another directory if necessary.
I agree that the error message is confusing though, I will see how we can change that to something more meaningful. Thanks for the feedback.

thomas-bc, 2023-02-08:
Actually, that option already exists with `-p <PATH>`

bocchino, 2023-02-08:
Sounds good. The behavior seems good when the tool is used correctly. Improving the error message would be helpful.

LeStarch, 2023-02-13:
We should spin the incorrect error message into a new ticket.

closed 2023-02-13 by LeStarch