nasa/fprime #1807: Update Drv::BlockDriverImpl to add telemetry in Sched port

timcanham, 2022-12-09:
| | |
|:---|:---|
|**_F´ Version_**| 3.x |
|**_Affected Component_**| `Drv/BlockDriver` |
---
## Feature Description

Add some telemetry channels that count bytes sent/received (the same for this trivial example) and add more comments describing the pattern.

## Rationale

This makes the example clearer to readers. Sometimes there is confusion like in issue #1805.

matt392code, 2024-12-28:
The Block Driver currently acts as both a clock source for rate groups and a packet loopback device. We'll add telemetry to track bytes sent/received when the Sched port is called. This will demonstrate the common pattern of using scheduled telemetry updates to avoid overwhelming the system.
We can implement this:
**First, we need to add telemetry channels in the FPP file for the component:**
```
# In Drv/BlockDriver/BlockDriver.fpp

@ Component for an F' scheduling test driver
active component BlockDriver {

  # Existing ports...

  @ Bytes sent through the driver
  telemetry BytesSent: U32 id 0

  @ Bytes received (looped back) through the driver 
  telemetry BytesReceived: U32 id 1

  # Existing ports continue...
}
```
**Then update the implementation in BlockDriverImpl.cpp:**
```
// In Drv/BlockDriver/BlockDriverImpl.cpp

namespace Drv {

  // Add member variables to track bytes
  U32 BlockDriverImpl::m_bytesSent = 0;
  U32 BlockDriverImpl::m_bytesReceived = 0;

  void BlockDriverImpl::init(const NATIVE_INT_TYPE instance) {
    BlockDriverComponentBase::init(instance);
  }

  void BlockDriverImpl::BufferIn_handler(
      const NATIVE_INT_TYPE portNum,
      Buffer& buffer
  ) {
    // Update byte counts
    const U32 size = buffer.getSize();
    m_bytesSent += size;
    m_bytesReceived += size; // Since this is a loopback

    // Existing loopback behavior
    this->BufferOut_out(0, buffer);
  }

  void BlockDriverImpl::Sched_handler(
      const NATIVE_INT_TYPE portNum,
      NATIVE_UINT_TYPE context
  ) {
    // Update telemetry channels periodically when scheduled
    this->tlmWrite_BytesSent(m_bytesSent);
    this->tlmWrite_BytesReceived(m_bytesReceived);
  }

} // namespace Drv
```
**Update the header file:**
```
// In Drv/BlockDriver/BlockDriverImpl.hpp

namespace Drv {

  class BlockDriverImpl: public BlockDriverComponentBase {
    public:
      // Existing declarations...

    PRIVATE:
      // Track total bytes sent/received
      static U32 m_bytesSent;
      static U32 m_bytesReceived;
  };

} // namespace Drv
```
**Add clarifying comments in the topology file:**
```
# In Ref/Top/topology.fpp

# Block driver connections
  # Clock source for rate groups
  blockDrv.CycleOut -> rateGroupDriverComp.CycleIn

  # Schedule periodic telemetry updates from block driver
  # This avoids overwhelming telemetry when processing high-rate data
  rateGroup3Comp.RateGroupMemberOut[2] -> blockDrv.Sched
  ```
This implementation:
a.  Adds two telemetry channels to track bytes sent/received
b.  Updates the counts when processing buffers
c.  Emits telemetry periodically via the Sched port rather than on every packet
d.  Includes clarifying comments about the scheduling pattern
The telemetry will now be updated at the rate of Rate Group 3, providing periodic statistics without overwhelming the system. This demonstrates the common F' pattern of using scheduled updates for driver statistics.

LeStarch, 2025-02-04:
Can we just replace block driver with a more representative timing system?  It is always a source of confusion.

timcanham, 2025-02-04:
Yes, that would be a good idea.

matt392code, 2025-02-04:
2 attachments:
**The FPP specifications defining:**
A new TimingComponent that handles only timing/scheduling
A separate LoopbackComponent for packet testing

**The C++ implementation of the TimingComponent**
**Key improvements over the original Block Driver:**
Clear Separation of Concerns:

- Timing functionality is now isolated in TimingComponent
- Loopback testing is handled by a dedicated LoopbackComponent
- Each component has focused telemetry and commands

**Enhanced Timing Features:**

- Configurable intervals for rate groups
- Jitter monitoring and statistics
- Enable/disable control for individual rate groups
- Clear error reporting and event logging

**Better Testing Support:**

- LoopbackComponent includes detailed statistics
- Configurable artificial delays for testing
- Clear error handling and reporting

**To migrate from the Block Driver:**

Update your topology file to use the new components:
```
instance timing: Drv.TimingComponent base id 0x100
instance loopback: Drv.LoopbackComponent base id 0x200

# Connect timing to rate groups
timing.CycleOut[0] -> rateGroup1.CycleIn
timing.CycleOut[1] -> rateGroup2.CycleIn
timing.CycleOut[2] -> rateGroup3.CycleIn
timing.CycleOut[3] -> rateGroup4.CycleIn
```

[timing-system.txt](https://github.com/user-attachments/files/18650645/timing-system.txt)

[timing-implementation.txt](https://github.com/user-attachments/files/18650647/timing-implementation.txt)

LeStarch, 2025-03-27:
Closing, as we will replace BlockDriver with a proper timer component.

closed 2025-03-27 by LeStarch