nasa/fprime #1826: Add CMake formatter

ThibFrgsGmz, 2022-12-22:
| | |
|:---|:---|
|**_F´ Version_**| latest | 
|**_Affected Component_**| None |
---
## Feature Description

Add [cmake-format](https://github.com/cheshirekow/cmake_format) to the project. See https://github.com/nasa/fprime/pull/1820 for more information.

## Rationale

To homogenize all CMake files according to the policy dictated in a `.cmake-format.yml` file.

LeStarch, 2022-12-22:
@thomas-bc what would your thoughts be at adding this to your `fprime-util format` command?  In-fact we could extrapolate:

```
if file_extension in <C++ extensions>:
    clang-format
if file_extension == .cmake or file_name = "CMakeLists.txt":
    cmake-format
if file_extension in <Py extensions>
    black
```

Thoughts?  We'd then have all our formatting in on place....

thomas-bc, 2022-12-23:
Yes that would work nicely if we want to be using cmake-format!
Although I will say, `black ./` is really simple enough to use already - and we would need some refactor to include that in `fprime-util format` since it currently requires to be ran in F’ (or a deployment) to find the .clang-format. Unless using the `-d` flag is good enough?

matt392code, 2025-01-04:
[format-util.py.txt](https://github.com/user-attachments/files/18306206/format-util.py.txt)
Proposed enhanced version of the `fprime-util format` command that integrates multiple formatters, including cmake-format. Here are the key features:

1. Unified Formatting Interface
- Supports multiple formatters in one command
- Handles C++ files (clang-format), CMake files (cmake-format), and Python files (black)
- Maintains existing functionality while adding new capabilities

2. Configuration Management
- Looks for .cmake-format.yml in the deployment directory or parent directories
- Preserves existing formatter configurations
- Allows for project-wide formatting standards

3. Flexible Usage
```bash
# Format current directory
fprime-util format

# Format specific files or directories
fprime-util format path/to/file.cpp path/to/directory

# Format with deployment context
fprime-util format -d /path/to/deployment
```

4. Error Handling
- Validates formatter tool availability
- Provides helpful installation instructions if tools are missing
- Reports formatting errors without stopping the entire process

The implementation follows F Prime's existing patterns while adding the requested cmake-format functionality. The code is modular and can be easily extended to support additional formatters in the future.

To integrate this into F Prime:

1. Install required dependencies:
```bash
pip install cmake-format black clang-format
```

2. Add a .cmake-format.yml file to define CMake formatting rules:
```yaml
format:
  line_width: 80
  tab_size: 4
  use_tabchars: false
  max_subgroups_hwrap: 2
  max_pargs_hwrap: 3
  dangle_parens: false
```

closed 2026-01-13 by thomas-bc