nasa/fprime #1831: Question about Apache-2.0 in space

tmechelke, 2023-01-04:
Excerpt from the Apache-2.0 license:
... each Contributor hereby grants to You
        a perpetual, worldwide ... copyright license ...
        a perpetual, worldwide ... patent license ...

The copyright and patent grants are restricted: **worldwide**.
A satellite around Earth is arguably not included by that definition.
Or is it? What about software on the Moon or on Mars?

Is Apache-2.0 licensed software legally usable in space?

astroesteban, 2023-01-04:
Do we have any space lawyers in the community? 👀

ThibFrgsGmz, 2023-01-04:
It would be good to ask this on cFS as well: they have lawyers in their process.

tmechelke, 2023-01-05:
I asked it over there as well: https://github.com/nasa/cFS/issues/637

ThibFrgsGmz, 2023-01-08:
Note: according to ChatGPT, the answer is yes: "the Apache-2.0 license allows the use of Apache licensed software in space". 😁

astroesteban, 2023-01-08:
Our AI overlords have spoken 🤖

LeStarch, 2023-01-17:
Given this was pinged over at CFS, I will close here for the time being.

closed 2023-01-17 by LeStarch