nasa/fprime #1840: Add GitHub Action - Markdown link check

ThibFrgsGmz, 2023-01-08:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**|  |
---
## Feature Description

A workflow for checking broken links in Markdown files should be integrated through the GitHub Action [markdown-link-check](https://github.com/gaurav-nelson/github-action-markdown-link-check).

This GitHub Action should check all Markdown files in the repository and report any broken links it detects.

The action should be configurable to exclude certain parts of the repository (e.g., specific directories or files) from link checking.

If broken links are detected, the action should display an error and stop the workflow, with a message indicating the files and links involved.

## Rationale

Broken links can be frustrating for users and affect the quality and user experience of a site or software. They can also be difficult to spot manually, especially in large repositories with many Markdown files.

Adding a GitHub Action that automatically checks for broken links in Markdown files would make it easier to maintain and improve the quality of the repository by quickly flagging broken links and allowing contributors to fix them.

It could also improve the user experience for people using the repository, providing them with quality content without broken links.

SauravMaheshkar, 2024-03-27:
I'd like to take this up 😃

thomas-bc, 2024-03-27:
That'd be great! I really would love to have this, just never got the time to get to it.

thomas-bc, 2024-03-27:
Once you have the PR ready for this, we can help fixing the markdown links that may be invalid. It's possible some might not be that trivial to resolve if files have been moved around and such.

PR #2651 (2024-04-24): feat(ci): add markdown link checker
| | |
|:---|:---|
|**_Related Issue(s)_**| #1840  |
|**_Has Unit Tests (y/n)_**| n |
|**_Documentation Included (y/n)_**| n |

---
## Change Description

This PR adds a workflow for checking broken links in Markdown files

## Rationale

Fixes #1840
closed 2024-04-24 by thomas-bc