nasa/fprime #1850: F' `fprime-util` Fails to generate build: "Failed to run fpp-locate-deps"

dmitri-mcguckin, 2023-01-16:
| | |
|:---|:---|
|**_F´ Version_**| 3.1.1 |
|**_Affected Component_**| Ref |
|**_Python Version_**| 3.10.9 |
|**_Pip Version_**| 22.3.1 |
|**_CMake Version_**| 3.25.1 |
---

## Problem Description

From what I've determined so far, there seems to be an  initial installation issue with FPrime v3.1.1. Occasionally this issue goes away and the install works for my current environment but when I return the next day to continue my work, the same error occurs again as if the build-state is dirtied.

The only temporary fix I've found so far is to fully blow away the virtual environment and re-install everything again, but even this has had inconsistent results. (This is in reference of the suggestions in Issue #1493)

After a fresh download, and isolated Python Virtual Environment, the standard command: `fprime-util generate --ut` fails with the following logs:

```
[INFO] Generating build directory at: ~/Downloads/fprime-3.1.1/Ref/build-fprime-automatic-native
[INFO] Using toolchain file None for platform default
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Searching for F prime modules in: ~/Downloads/fprime-3.1.1
-- Autocoder constants file: ~/Downloads/fprime-3.1.1/config/AcConstants.ini
-- Configuration header directory: ~/Downloads/fprime-3.1.1/config
-- [python3] python3 found at: ~/Downloads/fprime-3.1.1/.venv/bin/python3
-- [fpp-tools] fpp-depend found at: ~/Downloads/fprime-3.1.1/.venv/bin/fpp-depend
-- [fpp-tools] fpp-to-xml found at: ~/Downloads/fprime-3.1.1/.venv/bin/fpp-to-xml
-- [fpp-tools] fpp-to-cpp found at: ~/Downloads/fprime-3.1.1/.venv/bin/fpp-to-cpp
-- [fpp-tools] fpp-locate-defs found at: ~/Downloads/fprime-3.1.1/.venv/bin/fpp-locate-defs
-- Target build toolchain/platform: Linux/Linux
-- Including ~/Downloads/fprime-3.1.1/cmake/platform/Linux.cmake
-- Requiring thread library
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Installation directory: ~/Downloads/fprime-3.1.1/Ref/build-artifacts
-- Performing CMake source prescan
-- Performing CMake source prescan - DONE
-- Generating FPP location index
CMake Error at ~/Downloads/fprime-3.1.1/cmake/target/fpp_locs.cmake:56 (message):
-- Configuring incomplete, errors occurred!
  Failed to run fpp-locate-deps
See also "~/Downloads/fprime-3.1.1/Ref/build-fprime-automatic-native/CMakeFiles/CMakeOutput.log".
Call Stack (most recent call first):
  ~/Downloads/fprime-3.1.1/cmake/target/fpp_locs.cmake:95 (generate_locations)
  ~/Downloads/fprime-3.1.1/cmake/target/target.cmake:67 (fpp_locs_add_global_target)
  ~/Downloads/fprime-3.1.1/cmake/target/target.cmake:67 (cmake_language)
  ~/Downloads/fprime-3.1.1/cmake/API.cmake:465 (setup_global_target)
  ~/Downloads/fprime-3.1.1/cmake/API.cmake:426 (register_fprime_target_helper)
  ~/Downloads/fprime-3.1.1/cmake/FPrime.cmake:72 (register_fprime_target)
  CMakeLists.txt:32 (include)
```

The primary failure appears to be that `fprime-util` is attepmting to exec the `fpp-locate-deps` script, and after peering into my `./.venv/bin` directory, it is infact non-existent. The contents of my VEnv binary dir are:

```
Activate.ps1
activate
activate-global-python-argcomplete
activate.csh
activate.fish
chardetect
cheetah
cheetah-analyze
cheetah-compile
cookiecutter
flask
fpp-check
fpp-depend
fpp-filenames
fpp-format
fpp-from-xml
fpp-locate-defs
fpp-locate-uses
fpp-syntax
fpp-to-cpp
fpp-to-xml
fprime-cli
fprime-gds
fprime-seqgen
fprime-util
fprime-version-check
gcovr
markdown_py
normalizer
pip
pip3
pip3.10
py.test
pygmentize
pytest
python -> /usr/bin/python
python-argcomplete-check-easy-install-script
python-argcomplete-tcsh
python3 -> python
python3.10 -> python
register-python-argcomplete
slugify
v1.0.2
```

## How to Reproduce

1.  Download, decompress, and change dirs into the v3.1.1 source-code tarball from the [releases page](https://github.com/nasa/fprime/releases)
2. Setup the virtual environment: `python3 -m venv ./.venv && source ./.venv/bin/activate`
3. Install non-project dependencies: `pip install -U setuptools setuptools_scm wheel pip`
4. Install project dependencies: `pip install -r requirements.txt`
5. Change to `./Ref` directory and generate the project: `fprime-util generate --ut`

## Expected Behavior

I expect the fprime repository to generate correctly without error and create a new `build-fprime-automatic-native-ut` folder.

dmitri-mcguckin, 2023-01-17:
Quick update to this issue.

I've been diving deeper into what exactly is being called by CMake and now I realize that `fpp-locate-deps` is just an alais for [this cmake function](https://github.com/nasa/fprime/blob/master/cmake/target/fpp_locs.cmake#L47-L59) which is essentially a call to `autocoder/fpp-wrapper/fpp-redirect-helper` which is apparently failing for some reason.

dmitri-mcguckin, 2023-01-17:
One more update, I'm picking apart what this call to `fpp-redirect-helper` is supposed to look like

After adding:

```cmake
message("-- EXEC: `${FPP_LOCATE_DEFS_HELPER} ${FPP_LOCS_FILE}" "${FPP_LOCATE_DEFS} -d ${CMAKE_BINARY_DIR}`")
```

Before

```cmake
execute_process(COMMAND "${FPP_LOCATE_DEFS_HELPER}" "${FPP_LOCS_FILE}" "${FPP_LOCATE_DEFS}" -d "${CMAKE_BINARY_DIR}" ${FPP_CONFIGS} ${FPP_FILES} RESULT_VARIABLE result)
```

I got:

```
fprime/cmake/autocoder/fpp-wrapper/fpp-redirect-helper fprime/Ref/build-fprime-automatic-native-ut/locs.fppfprime/.venv/bin/fpp-locate-defs -d fprime/Ref/build-fprime-automatic-native-ut
```

So I assume from this, a generic command is supposed to look something like this from within the fprime repo?

```bash
exec cmake/autocoder/fpp-wrapper/fpp-redirect-helper \
    Ref/build-fprime-automatic-native-ut/locs.fpp \
    .venv/bin/fpp-locate-defs \
    .venv/bin/fpp-locate-defs \
    -d Ref/build-fprime-automatic-native-ut
```

dmitri-mcguckin, 2023-01-17:
Well, surprise, surprise, it was a permissions issue.

After a bit more tinkering and finagling with environment variables, I ran the above command manually and got a `Permission Denied` which is what cause the target in CMake to fail as well.

I do have some tools that were trying to fix the permissions of files it doesn't recognize, and since some of the CMake helper scripts are bare files with the `#!` shebang instead of `*.sh` files, my tools assumed they were just plain text files and removed the executable permissions.

I've done a quick fix for this in my fork by simply adding `*.sh` to all of those files and then creating a relative soft link per-directory to do `helper_script -> helper_script.sh`.

***

&nbsp;

From here I'm open to suggestions as to what is apropos.

At-the-least we may want to add this to the troubleshooting page.

Another idea may be to append the file type to all of the shell scripts so they don't get mis-classified?

Do let me know your guys' thoughts.

LeStarch, 2023-01-17:
Typically these files are checked in with the correct permissions set on the helper scripts with-in the repo.  Is there something in your environment that is removing, or clearing those permissions?

dmitri-mcguckin, 2023-01-17:
Yes, as described above, I have tools that attempt to clean up permissions for files in my home or other sensitive directories such that it automatically removes the executable flag for anything it doesn't detect as an executable.

That much is an environment conflict that's local to me and am willing to fix that locally.

In context of F' though, the part that threw me for a loop was that CMake normally displays some error message right before the generic "CMake failed" at the end.

I'm wondering if we could either document this in the troubleshooting section and/or modify `execute_process` in `fpp_locs.cmake` to always redirect the output of the command to stdout. That way, at the least, the user will know if there's some niche problem on their filesystem via the logs.

LeStarch, 2023-01-18:
Ah, I understand.  Yes, we should find a way to expose these errors correctly.  I see you made an issue above to track that, thank you!

It is most certainly a deficit as it isn't reporting useful errors.

dmitri-mcguckin, 2023-01-23:
Closed via PR #1854

closed 2023-01-23 by dmitri-mcguckin