nasa/fprime #1853: Use of the "Conventional Commit" standard for project commits

ThibFrgsGmz, 2023-01-17:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**|  |
---
## Feature Description

Adding the [Conventional Commit](https://www.conventionalcommits.org/en/v1.0.0/) standard to all commits made to the project will allow for better readability and understanding of the changes made to the code.

Commits should be formatted as follows: `<type>(<scope>): <subject>`.

For example: `feat(module): added trajectory calculation functionality` or `fix(module): fixed a bug in satellite communication management`.

To help the adoption of this standard, a series of steps can be implemented:

- Setting up a pre-commit tool to check that contributors comply with the standard before they can commit.
- Add a section in the project documentation explaining the standard and the types of commits expected.
- Organize a training session for team members to help them understand the usefulness of the standard and how to use it.

Using the Conventional Commit standard, it is also possible to automatically generate detailed changelogs using tools such as [conventional-changelog](https://www.npmjs.com/package/conventional-changelog) or [commitlint-config-conventional](https://www.npmjs.com/package/@commitlint/config-conventional).
This makes it easier for end users and contributors to write release notes, providing a clear overview of the changes made to the project.

## Rationale

The "Conventional Commit" standard has become an industry standard for software commits. It allows :

- a better understanding of the modifications made to the code by specifying the type of commit (new functionality, bug fix, documentation, ...)
- a better traceability by allowing to follow the history of the commits and to understand the motivations behind the modifications.
- a better collaboration between team members by allowing to quickly understand the modifications made by the different contributors.

Automatic generation of detailed changelogs to facilitate communication of changes to the project.

The implementation of this standard can however take some time and may seem restrictive for contributors in the short term, but in the long term it brings gains in productivity, quality and maintainability that are worth the effort.

astroesteban, 2023-01-18:
I like this standard but I would suggest we select a subset of the possible scopes (feat, fix, docs, etc). Just to make things simpler for contributors maybe we use only feat and fix? What are y'all's thoughts on this?

ThibFrgsGmz, 2023-01-23:
Personally, I prefer to follow the entire standard, even though it seems rather cluttered and little or not used at all. As this standard becomes more and more common in the industry, contributors who are familiar with it won't feel limited if they can use it fully.

Or why not try using [Gitmoji](https://gitmoji.dev/) ? Maybe it's my hipster side talking, but I find it quite original and I seem to be seeing it more and more often. There are many tools that come with it (see [here](https://gitmoji.dev/related-tools)):
- [gitmoji-vscode](https://github.com/seatonjiang/gitmoji-vscode) : Gitmoji tool for git commit messages in VS Code
- [gitmoji-changelog](https://github.com/frinyvonnick/gitmoji-changelog/) : A changelog generator for gitmoji.
- [gitmoji-changelog-action](https://github.com/sercanuste/gitmoji-changelog-action) : GitHub action for gitmoji-changelog

LeStarch, 2023-01-23:
So to drop in some thoughts:

1. I would not ask committers to do this.  It is too large a group to control and it would provide a substantial barrier if committers needed to rewrite all their commits.
2. We **can** apply it to the people who squash-and-merge.  That would make the F´ history contain these well formatted commit messages.  Thus item 1 would only apply to sub-commit comments and have less of a direct impact on the commit log.
3. While I do like gitmoji it was quit jarring to see it the first time.  I think it is a bit much for our project given the range of customers we have today.

thomas-bc, 2024-06-04:
Converting this issue to a discussion as it fits the bill more