nasa/fprime #1860: Adoption of a time-based Release Process

ThibFrgsGmz, 2023-01-24:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**| Release Process |
---
## Feature Description

The project does not currently release versions regularly, despite the changes that have been made, which can lead to inconsistencies for users and contributors.

For example, this can cause coordination and planning problems, as they do not know when the changes they have made will be released, and so cannot plan their work effectively. It can also lead to conflicts between the different changes made, as they are not integrated on a regular basis.

I suggest adopting a [release process similar to Zephyr's](https://docs.zephyrproject.org/latest/project/release_process.html#release-process), based on a time cycle that balances the latest technologies and features with the overall quality of the release.

This process is based on the Linux kernel release model, with a development phase, followed by a stabilisation phase, and a release candidate tag for analysis and approval of the version before release.

## Rationale

This time-based release process will provide users with a stable and reliable version with the latest features.

It will also allow contributors to better plan their work and ensure that changes are released regularly.

Finally, using a release process similar to that of other large open source projects such as Zephyr will demonstrate that our project is in line with industry best practice.

ThibFrgsGmz, 2023-02-28:
To support this claim, here is a lesson learned from the Linux community available in a public report from the Linux Foundation (https://storage.pardot.com/6342/188781/Publication_LinuxKernelReport_2017.pdf p.25):

"Short release cycles are important.
In the early days of the Linux project, a new major kernel release
only came once every few years. That meant considerable delays in
getting new features to users, which was frustrating to users and
distributors alike. But, more importantly, such long cycles meant that
huge amounts of code had to be integrated at once, and that there
was a great deal of pressure to get code into the next release, even if
it wasn’t ready.

Short cycles address all of these problems. New code is quickly made
available in a stable release. Integrating new code on a nearly
constant basis makes it possible to bring in even fundamental
changes with minimal disruption. And developers know that if they
miss one release cycle, there will be another one in two months, so
there is little incentive to try to merge code prematurely."

LeStarch, 2023-02-28:
In general, I support this concept as more frequent releases would be a benefit to everyone.   This  would require a deeper restructuring of how F´ does development.  

Thus we need to come up with a plan on how we might implement this.

LeStarch, 2023-03-01:
After some discussion we've decided we'd like to move to the following model:

1. Features and general development: when a release's development is finished.
2. Bugs and maintenance: quarterly release

We are still working on the details and how this will fall out, but that is where we'd like to target.

closed 2023-05-03 by LeStarch

thomas-bc, 2023-05-03:
We are moving forward with this! See our [milestones](https://github.com/nasa/fprime/milestones) 🚀 
We will populate each milestone with more items, coming soon.

ThibFrgsGmz, 2023-05-03:
Sounds awesome guys!