nasa/fprime #1862: Code coverage analysis does not work as expected for autocoded port C++

bocchino, 2023-01-24:
When testing code generation for ports, the F Prime build system does not provide coverage analysis for the generated port code. See https://github.com/nasa/fprime/pull/1852.

bocchino, 2023-04-19:
@LeStarch What is the current status of this issue? I recall that we discussed a potential fix.

LeStarch, 2023-04-19:
I believe you can use flags to specify how coverage should treat autocode....so it should be closed.

```
fprime-util check --coverage --port-ac
```

bocchino, 2023-04-19:
Thanks! @tiffany1618 Can you try this and see if it works for your use case?

tiffany1618, 2023-04-25:
@bocchino @LeStarch Unfortunately `--port-ac` and `--comp-ac` don't seem to be working for me in FppTest project (`--type-ac` works as expected). The correct files appear in the coverage report, but with 0% coverage. In the build cache, the .gcda files are missing.

bocchino, 2023-04-25:
I am able to reproduce the behavior that Tiffany described on the head of devel as follows:

1. In `FppTest/port`, run `fprime-util generate --ut` and then `fprime-util check --coverage --port-ac --comp-ac --type-ac`.
2. Open `coverage/coverage.html`. Most of the files appear with 0% coverage. See below. We know from previous analysis that these files have high coverage.

[image]

LeStarch, 2023-04-26:
`--type-ac` works as expected: does this mean that it produces the expected coverage, or that the files appear with the 0% coverage?

tiffany1618, 2023-04-26:
I've actually noticed some inconsistencies upon further testing.

Procedure:

1. In `FppTest`, run `fprime-util generate --ut`.
2. In the specified directory, run `fprime-util check --coverage <coverage-flag>`.
3. In the specified directory, open `coverage/coverage.html`.

`--type-ac`
- In `FppTest/enum`, `FppTest/array`, `FppTest/struct`: correct files and coverage
- In `FppTest/port`: correct files but 0% coverage

`--port-ac`
- In `FppTest/port`: correct files but 0% coverage

`--component-ac`
- In `FppTest/port`: correct files and coverage
- In `FppTest/component/active` on [the FPP component integration branch](https://github.com/tiffany1618/fprime/tree/fpp-component-integration): correct files but 0% coverage

PR #2013 (2023-05-18): Fix missing coverage compilations for unit-test files
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

This fixes #1862.  The issue was unit-tests were not being compiled with coverage flags, meaning the profiling was incomplete.