nasa/fprime #1867: Add command parameter validation

timcanham, 2023-01-25:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Add command parameter ranges and validation to FPP, component AC, and sequence compiler.

## Rationale

A rationale for why the feature should be implemented in F´.

timcanham, 2023-07-05:
@bocchino Did this ever make it into FPP?

bocchino, 2023-07-13:
It's on the to-do list. We are tracking it here: https://github.com/fprime-community/fpp/issues/127.

matt392code, 2025-01-07:
[fprime-param-impl.txt](https://github.com/user-attachments/files/18328241/fprime-param-impl.txt)
[fprime-param-ranges.txt](https://github.com/user-attachments/files/18328237/fprime-param-ranges.txt)

Proposed implemented the parameter range validation across three key areas:

1. **FPP Range Definitions**:
   - Added range annotations for numeric types
   - Defined enumerated ranges for operating modes
   - Created custom types with built-in range validation

2. **Component AC Implementation**:
   - Added parameter validation in command handlers
   - Implemented validation error events
   - Created internal validation port for reusability

3. **Sequence Compiler Integration**:
   - Added pre-compile time parameter validation
   - Implemented range checking in sequences
   - Added support for conditional validation

Key features:
- Type-safe range validation
- Compile-time checks where possible
- Runtime validation with proper error handling
- Custom validation support
- Clear error reporting

bocchino, 2025-01-07:
Closing this issue as a duplicate.

closed 2025-01-07 by bocchino