nasa/fprime #1868: Sequence Linter

timcanham, 2023-01-25:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

A quick description of the requested feature.

## Rationale

A rationale for why the feature should be implemented in F´.

LeStarch, 2024-03-28:
@timcanham another placeholder.  Can we elaborate or close?

matt392code, 2025-01-07:
[fprime-sequence-linter.py.txt](https://github.com/user-attachments/files/18328283/fprime-sequence-linter.py.txt)
Proposed basic sequence linter framework for F Prime that performs several key validations. Here are the main features:

1. Validates basic sequence format and structure
2. Checks command timing and ordering
3. Identifies potential resource conflicts
4. Verifies command dependencies
5. Validates command parameters

The linter can be configured via a JSON configuration file and supports both text and JSON output formats.

Some of the placeholder methods need to be implemented with F Prime-specific logic:
- Command format validation
- Timestamp extraction
- Resource usage tracking
- Parameter validation
- Dependency checking

LeStarch, 2025-02-04:
Closing as no-elaboration provided.

closed 2025-02-04 by LeStarch