nasa/fprime #1893: AcConstants.ini exists but is not used

bocchino, 2023-02-15:
## Issue

* AcConstants.ini exists in the repo but isn't used for anything: https://github.com/nasa/fprime/blob/devel/config/AcConstants.ini. It was used in the hand-coded XML, which no longer exists. It has been superseded by AcConstants.fpp. This is confusing because users may update AcConstants.ini by mistake, when they should update AcConstants.fpp.
* The docs still refer to AcConstants.ini. See https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/dev/configuring-fprime.md#acconstantsini.

## Suggested Solution

1. Remove AcConstants.ini from the repo.
2. Update the docs.

bocchino, 2023-02-15:
@gjwatney FYI.

LeStarch, 2023-02-15:
@bocchino is this something you would fix?  It seems fairly straight-forward.

bocchino, 2023-02-16:
This also needs to be fixed:

https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/user/settings.md

Also, it looks like something in the F Prime tools is using settings.ini to find AcConstants.ini.

bocchino, 2023-02-16:
There are also references to AcConstants.ini here and there in the docs, e.g.:

https://github.com/nasa/fprime/blob/devel/Svc/StaticMemory/docs/sdd.md#configuration

We should update these to refer to AcConstants.fpp.

bocchino, 2023-02-16:
> Also, it looks like something in the F Prime tools is using settings.ini to find AcConstants.ini. This should be scrubbed.

Actually I think this is the first order of business. Otherwise it is not clear what will happen if the AcConstants.ini file is not present, and the tools are expecting it to be there.

LeStarch, 2023-03-01:
@bocchino doing this change will break our backward support of the XML modeling language.  Have we officially drop that support?

bocchino, 2023-03-01:
> doing this change will break our backward support of the XML modeling language

Can you elaborate the use case you have in mind that will break? AcConstants.ini defines some symbols that were used by the old handwritten XML and are not used by the FPP-generated XML. So at a minimum we should delete all those unused symbols and document that AcConstants.ini is there for backwards compatibility. Are you thinking that removing AcConstants.ini entirely will break something that users are doing with their own handwritten XML? That seems unlikely to me, since the purpose of AcConstants.ini was to configure the framework, and we are not using it for that (or for any other purpose) any more.

bocchino, 2023-03-01:
After discussion, hand-writers of XML components might want to use AcConstants.ini to configure their XML. However, we will begin the process of phasing out support for hand-written XML components.

closed 2024-02-02 by thomas-bc