nasa/fprime #1894: Add --pass-through arguments to `fprime-gds`

LeStarch, 2023-02-15:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Allow fprime-gds to pass through arguments to FSW that is automatically started up.

thomas-bc, 2024-02-21:
This would allow for example
```
fprime-gds --pass-through -p 50000 -a 127.0.0.1 --other-flags
```

Implementing and parsing pass-through can be a little tricky, here's how we have done it in `fprime-util` : https://github.com/fprime-community/fprime-tools/blob/dff8b670f01eea41cf5ed76f63825c2d07a390dd/src/fprime/util/cli.py#L222-L227

matt392code, 2025-01-07:
[fprime-cli-mod.py.txt](https://github.com/user-attachments/files/18328345/fprime-cli-mod.py.txt)
Proposed implementation does the following:

1. Adds a `--pass-through` argument to the GDS parser following the same pattern used in the format parser. This uses `argparse.REMAINDER` to capture all remaining arguments after `--pass-through`.

2. Modifies the GDS command handler (`run_gds`) to:
   - Check for pass-through arguments using `hasattr(parsed, "pass_through")`
   - Extend the FSW arguments list with any pass-through arguments
   - Pass these arguments to the FSW binary via the `launch_process` function

This implementation allows users to run commands like:
```bash
fprime-gds --gui --pass-through -p 50000 -a 127.0.0.1 --other-flags
```

The arguments after `--pass-through` will be passed directly to the FSW executable that GDS launches.

Some key points about this implementation:
1. `--pass-through` must be the last argument since it captures all remaining arguments
2. The implementation follows the existing pattern used in the format parser
3. Arguments are passed through to the FSW binary unmodified
4. The feature integrates with the existing GDS launch process

closed 2025-10-01 by thomas-bc