nasa/fprime #1918: Add directory listing to `Svc/FileManager`

timcanham, 2023-03-01:
| | |
|:---|:---|
|**_F´ Version_**| 3.x |
|**_Affected Component_**| `Svc/FileManager` |
---
## Feature Description

Add a command that will list the files in a directory with file sizes.

## Rationale

See what files are currently on the deployment.

LeStarch, 2024-03-28:
@timcanham we have this....we just want to remove the CRC so that we don't have timing issues.  Or am I missing something?

Joshua-Anderson, 2024-03-28:
Now that we (nearly!) have data products, we should have a file listing command that generates a file listing data product.

The current command generates one EVR per listed file - this can be problematic because EVRs can get dropped or delayed, leading to an inconsistant view of the file system. Including an entire file listing in one data product provides a more consistant view of filesystem state.

That being said - you also may need to bootstrap your system if you're having filesystem issues, so there's value to having both an EVR and a data product based file listing command.

timcanham, 2024-03-28:
@Joshua-Anderson FYI, the upcoming `DpCatalog` component will generate a DP that will have things like size, priority, state (transmitted or not)...

matt392code, 2025-01-07:
[file-manager.txt](https://github.com/user-attachments/files/18328403/file-manager.txt)
Proposed implementation that includes both EVR and data product methods for directory listing. Here are the key features:
1. **Dual-Mode Operation**:
   - The command handler takes a boolean flag `useDataProduct` to select the method
   - Both implementations live in separate methods for clarity
   - Either approach can be used depending on system state/needs

2. **EVR-based Implementation** (`listDirWithEvr`):
   - Emits one EVR per file entry
   - Lightweight and good for bootstrapping
   - No buffer limitations
   - Real-time file-by-file reporting

3. **Data Product Implementation** (`listDirWithDataProduct`):
   - Collects all entries first
   - Creates a single serialized data product
   - Includes a summary EVR at completion
   - Provides atomic view of directory state

4. **Safety Features**:
   - Entry buffer size limit (1000 entries)
   - Skips "." and ".." directory entries
   - Error handling for directory operations
   - Serialization status checking

To use this in F Prime, you would:

1. Add this component to your topology
2. Define the command and data product XML
3. Generate the Ac.hpp/cpp files
4. Include appropriate unit tests

LeStarch, 2025-02-04:
Is this still and issue, and what is the right approach here?

Joshua-Anderson, 2025-02-06:
I think this is still an issue - it would be awesome if F' supported two types file listing commands, and I don't think either are currently supported in core F'

1. A simple, non-filesystem dependent EVR-based file listing that can be used to debug filesystem issues.
2. A file listing that generates a data product, that provides a more efficient and easier to work with snapshot of filesystem status (since individual EVRs get dropped it can be hard to determine if a EVR-based file listing is complete).

closed 2025-09-10 by thomas-bc