nasa/fprime #1927: Broken link in GPS tutorial

keck-in-space, 2023-03-07:
| | |
|:---|:---|
|**_F´ Version_**| na |
|**_Affected Component_**| documentation   |
---
## Problem Description

There is a broken link in the GPS tutorial. It links to a User's Guide PDF which does not exist.

https://github.com/nasa/fprime/blob/7bf79db61b1b63d60a624328dc909ee62869e8ba/docs/Tutorials/GpsTutorial/Tutorial.md?plain=1#L60

AngeloSorte, 2023-03-20:
Hi. I am new to contributing here. I don't understand well how this should be solved, but I found a link to the guide here on GitHub, on someone else's page. Can it be useful?

AngeloSorte, 2023-03-20:
> The updated location for the user's guide is here: https://nasa.github.io/fprime/UsersGuide/guide.html

Ah, ok. I though a PDF version was needed.

LeStarch, 2023-05-03:
We have scrubbed the GPS tutorial due to it not being updated with FPP.  Closing this issue.

closed 2023-05-03 by LeStarch