nasa/fprime #1931: CI CppCheck/CppScan failing

thomas-bc, 2023-03-13:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

See https://github.com/nasa/fprime/actions/runs/4390265240/jobs/7688586298. CI is failing on CppLint and CppCheck scans because the `--force` option of Microsoft's SARIF-Multitool converter is not available. Probably an update of the convert tool.

Fix is likely going to be to simply remove the option, but we should also check what it used to be doing to be sure we're not losing any functionality.

thomas-bc, 2023-03-13:
@ThibFrgsGmz do you want to take a look into that or should I?

ThibFrgsGmz, 2023-03-13:
@thomas-bc I'll take a look tonight. While browsing the github repo I see some people get the same issue: https://github.com/microsoft/sarif-sdk/issues/2638

ThibFrgsGmz, 2023-03-14:
For the time being, we may revert to the previous version, i.e. 4.0.0, by replacing the installation commands:

https://github.com/nasa/fprime/blob/bbf049ecbd82895a0c2de3e5d126dcea55618ceb/.github/workflows/cpplint-scan.yml#L33

and 

https://github.com/nasa/fprime/blob/bbf049ecbd82895a0c2de3e5d126dcea55618ceb/.github/workflows/cppcheck-scan.yml#L32
 with `npm i @microsoft/sarif-multitool@4.0.0`

thomas-bc, 2023-03-16:
Fixed - Thanks for working on this!

PR #1932 (2023-03-14): Fix #1931 - `microsoft/sarif-multitool` usage changes
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fix #1931 

## Rationale

See ticket description

## Testing/Review Recommendations

Investigate CI logs

## Future Work

None
closed 2023-03-16 by thomas-bc