nasa/fprime #1936: Insufficient permissions to set task priority warning

pcrosemurgy, 2023-03-17:
| | |
|:---|:---|
|**_F´ Version_**| [91c578e](https://github.com/nasa/fprime/commit/91c578ee37c01ba92e556dbca0287d82f9f18b95) (latest `devel` commit) |
|**_Affected Component_**|  Task thread priorities |
---
## Problem Description

When running the F' system reference deployment on stable (Bullseye) Debian there are "Insufficient Permissions" warnings related to task threads. There's also an F'-specific message warning developers that future releases will fail under these conditions.

`uname -a` reports:
```
Linux e29ca9579ba8 5.15.49-linuxkit #1 SMP Tue Sep 13 07:51:46 UTC 2022 x86_64 GNU/Linux
```

Upon starting a deployment each thread reports the following warning:
```
[WARNING] High task priority of 140 being clamped to 99
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_blockDrv. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[WARNING]      Note: future releases of fprime will fail when setting priority/affinity without sufficient permissions

[WARNING] Task priority set and permissions unavailable. Discarding priority.
[WARNING] High task priority of 100 being clamped to 99
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_camera. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[WARNING]      Note: future releases of fprime will fail when setting priority/affinity without sufficient permissions

[WARNING] Task priority set and permissions unavailable. Discarding priority.
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_chanTlm. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[WARNING]      Note: future releases of fprime will fail when setting priority/affinity without sufficient permissions

...
```

While these are currently just warnings it sounds like this will eventually be fatal. I am also curious if it is possible to properly set these "permissions" when running on Debian?

## How to Reproduce

This can be reproduced by compiling and running the F' system reference with the following steps:
1. `fprime-util generate`
2. `fprime-util build`
3. `build-artifacts/Linux/bin/Ref`
4. See warnings above

## Expected Behavior

Maybe this is expected under Debian but if so then I think the `Note: future releases of fprime will fail ...` note should likely be changed?

pcrosemurgy, 2023-03-17:
I believe @kevin-f-ortega has seen these warnings as well.

pcrosemurgy, 2023-03-20:
@LeStarch In the example above I was running as root in a Debian container.

LeStarch, 2023-03-20:
It may be the case that the container itself cannot set thread priority, I am not entirely sure.