nasa/fprime #1943: `fprime-util impl` should generate fully qualified header guards

pcrosemurgy, 2023-03-23:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

While working on a custom `Framer.fpp` to replace `Svc.Framer` I have to hand-edit the autogenerated `Framer.hpp-template` header guards to be fully-qualified to avoid accidentally `#ifndef`'ing too liberally.

For example, `Framer.fpp` within `module Foo` generates:
```cpp
#ifndef Framer_HPP
#define Framer_HPP
...
#endif 
```

It may be best to instead generate:
```cpp
#ifndef Foo_Framer_HPP
#define Foo_Framer_HPP
...
#endif 
```

This is a bit of a mix between a feature request / bug report. It's somewhat reasonable to expect hand-editing these `#ifndef` if guards but if this can be autogenerated safely to begin with that would be a nice-to-have.

LeStarch, 2023-05-03:
I thought we merged a fix for this.  Did we not?

thomas-bc, 2023-05-03:
@LeStarch I don't remember seeing that go through, but I may have missed it?

pcrosemurgy, 2023-05-03:
@LeStarch is referring to #1935, which I think should close this. Sorry for the confusion, I can't seem to replicate this so I think I made this ticket while mistakenly not using an F-Prime with #1935 merged.

closed 2023-05-03 by pcrosemurgy