nasa/fprime #1964: Sanitization Caused Unit Test Segfault on Mac OS

Joshua-Anderson, 2023-04-16:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**|  build system |
---
## Problem Description

On the development version of `fprime-util`, sanitization is enabled in unit tests builds by default.

On Mac OS with sanitization enabled, unit tests were crashing with a SEGFAULT (EXC_BAD_ACCESS) on initalization.

## How to Reproduce

1. Build and execute until tests with sanitizers enabled - unit test executables segfault

## Expected Behavior

Sanitization shouldn't crash unit tests.

Joshua-Anderson, 2023-04-16:
@thomas-bc I suspect this is a linking flag issue with the sanitizers, I know that they can be a bit finicky on macos when I've set them up previously.

Joshua-Anderson, 2023-04-16:
We also may want to start running CI against the alpha releases of fprime-util to catch issues like this more quickly

thomas-bc, 2023-04-17:
@Joshua-Anderson can you tell me a little more about your setup and what you're testing so I can try to reproduce? I have not had this issue show up before, and I'm able to build and run UTs (at least for the `Ref/` app) on MacOS with sanitizers enabled through fprime-util.
The CI is [running against](https://github.com/nasa/fprime/blob/91c578ee37c01ba92e556dbca0287d82f9f18b95/.github/actions/setup/action.yml#L14) what's in the requirements.txt, which is currently the latest alpha release.

Joshua-Anderson, 2023-04-17:
Ah - I was looking at the master branch, not devel. Glad to see we're testing against the alpha releases! Seems like I might have a somewhat more special case than I realized.

My machine:

- Intel Macbook with MacOS 13.3
- Apple clang version 12.0.0 (clang-1200.0.32.29

With an existing component, I ran `fprime-util generate --ut`, then `fprime-util impl --ut`, then copied the skeleton files to `test/ut` and added them to `CMakeLists.txt`

Running `fprime-util check` would crash with a segfault, and the backtrace showed the unit test executable crashing in dynamic linking `dylib`.

Purging and regenerating fprime-util generate --ut --disable-sanitizers` and no other changes caused the unit test executable to run as expected.

Given the only things the binary is dynamically linked against is the stdlib and the sanitization library, the seems related to sanitization library dynamic linking.

Joshua-Anderson, 2023-04-18:
Turns out upgrading my version of macOS was not updating the xcode command line versions and I was using a 2 year old version of clang that shipped with macOS 10.15 (catalina).

Forcing a reinstall of the tools upgraded me to clang 14, which worked as expected. This seems specifically like a clang 12 bug.

Given mavOS 10.15 is deprecated and this issue doesn't persist on modern versions of macOS, let's close this issue.

closed 2023-04-18 by Joshua-Anderson