nasa/fprime #1970: File Downlink problem crashes the flight software

gjwatney, 2023-04-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

If you donwnlink a file and it fails, for example due to permissions issues on the host, the GDS green dot will change to a red x and not recover.

## How to Reproduce
1. Have another user downlink a file so that /tmp/fprime-downlink is only writable by that user
2. Attempt file downlink as a new user
3. Observe red x in GDS window

## Expected Behavior
Generate an EVR that reports the permission issue and keeps running
Can fprime-gds be configured to either (1) make its directories group writable by default or (2) use a user's home directory for downlink? Otherwise, we all need sudo to keep changing permissions depending on who is downlinking

aggelosmots, 2023-04-20:
Have you tried running GDS in sudo?

LeStarch, 2023-04-20:
The issue is that when the GDS encounters a permissions issue, it shouldn't just crash.  It should recover and keep running.

gjwatney, 2023-05-05:
This is fixed in the following pull request
https://github.com/fprime-community/fprime-gds/pull/128

closed 2023-11-16 by thomas-bc