nasa/fprime #1981: Alternat `fprime-util new --deployment`

LeStarch, 2023-04-23:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Allow users to specify alternate base deployments.  

e.g. `fprime-util new --deployment https://github.com/fprime-community/fprime-baremetal-base-deployment.git`

thomas-bc, 2023-05-18:
@LeStarch how do you envision this functionality to be leveraged/used? Or rather, what would a `https://github.com/fprime-community/fprime-baremetal-base-deployment.git` look like? 
- is it on the user to make it a full-blown cookiecutter template?
- is it simply a deployment, that would basically just be cloned into the project? (this could work because `new --deployment` really doesn't take much configuration)
- is it a functional fprime project, and the utility needs to identify, extract and clone the deployment?
- is it something else I'm missing?

LeStarch, 2023-05-18:
@thomas-bc the baseline thought was to allow alternate cookie-cutter templates.  This is difficult for users....but is "a solution"  and easy for us to implement. It is also very flexible.

However, I am open to exploring other solutions too.

closed 2023-05-30 by thomas-bc