nasa/fprime #1984: Format C++ codebase

thomas-bc, 2023-04-25:
| | |
|:---|:---|
|**_F´ Version_**| `devel` |
|**_Affected Component_**| N/A |
---
## Feature Description

We are going to start formatting the entire C++ codebase and enforce the formatting through a CI check.

## Plan

- Create CI workflow to check that C++ is formatted correctly, with an optional exclude list. Start by excluding everything.
- Iteratively format the C++ codebase, removing the exclusion parameter as we go.

## Subtasks

- [ ] Create clang-format check workflow
- [ ] Format each top-level folder within F´

JohanBertrand, 2024-05-23:
Is anyone working on that right now? I'm happy having a look at it

thomas-bc, 2024-05-28:
@JohanBertrand if you don't mind I'll take the CI part of this ticket, because we have an idea of how we want to do it.

Once the check is there, feel free to help out formatting the code base!

We will format by small chunks and activate the CI per-module as we go.

thomas-bc, 2024-05-28:
One issue that is arising, and is loosely related to https://github.com/nasa/fprime-tools/pull/199

`clang-format`, for some obscure reason, has some slight variations in the formatting it outputs across different versions of the tool (even when using the exact same style). I have seen it happen with indentation of comments, and spacing like this one: https://github.com/thomas-bc/fprime/commit/a4585cde58301c0e201e797ea7a3ad9a7a090125

Some conversations about it [here](https://stackoverflow.com/questions/59395507/how-to-maintain-a-clang-format-file-for-different-clang-format-versions) and [here](https://stackoverflow.com/questions/58450914/unify-output-with-different-clang-format-versions)

We are likely going to need to pick a version of the tool we want to use in CI, version that users will need to use if they need to pass CI. It is very inconvenient... but there doesn't seem to be a way around it.

Let me know if you can think of anything else

cc @LeStarch @JohanBertrand

JohanBertrand, 2024-05-28:
I think it's reasonable to fix the version of the formatter, like it is also done with the static analyser (`clang-tidy-12`). Worst case scenario, the minor differences between two formatted versions would be flagged by the CI, and it could be resolved in one single commit.

It might be good to select a version of `clang-tidy` which is the same as `clang-format` to make it easier for the developers.

I'm not sure which version to select however. `clang-tidy-18` seem to have some new warnings and is detecting two new issues in the current configuration on the development branch. `clang-format-14` is adding the absolute path for the style that we discussed in https://github.com/nasa/fprime-tools/pull/199, which might be great to keep. The space difference that you mentioned as an example was introduced between `clang-format-17` and `clang-format-18`. 

Clang tidy/format 12 to 18 are easily available on ubuntu20.04 and 22.04 (I'm not sure about Mac OS), so any of those version would work for me.

For `fprime-util format`, we will need to decide if we want it to fail when the version is different than the one we expect. I guess we could send a warning during the formatting but still process the files if the version is different. What is your opinion?

matt392code, 2025-01-07:
[format-check-workflow.txt](https://github.com/user-attachments/files/18328600/format-check-workflow.txt)
Proposed GitHub Actions workflow for the C++ formatting check using clang-format. We'll need to:
1. Run clang-format checks on C++ files
2. Support an exclusion list
3. Use a specific clang-format version
4. Only check modified files to keep CI runs efficient
GitHub Actions workflow:
1. Triggers on PR and push to devel branch when C++ files or relevant config files change
2. Uses clang-format-14 (can be changed via CLANG_FORMAT_VERSION env var)
3. Only checks files that have changed in the PR/push
4. Supports an exclusion list via excluded_paths.txt
5. Provides clear output about which files need formatting

Key features:
- Uses actions/checkout to get the code
- Uses tj-actions/changed-files to efficiently get changed files
- Implements a Python script for checking formatting that:
  - Respects the exclusion list
  - Compares clang-format output with current content
  - Provides clear pass/fail status for each file
  - Exits with error if any files need formatting
- Uses .clang-format file in repo root for style configuration

To use this:
1. Place the workflow file in .github/workflows/clang-format-check.yml
2. Create/update .clang-format in repo root
3. Update excluded_paths.txt with paths to exclude
4. Optionally adjust CLANG_FORMAT_VERSION if needed

matt392code, 2025-01-07:
[format-folders.py.txt](https://github.com/user-attachments/files/18328617/format-folders.py.txt)
Proposed Python script that helps manage the folder-by-folder formatting process. Key features:
1. **Folder Management**:
   - Identifies top-level folders containing C++ files
   - Counts files by extension (.cpp, .hpp, .inl)
   - Tracks formatting progress in a state file

2. **Formatting**:
   - Supports both checking and applying formatting
   - Uses specified clang-format version
   - Handles errors gracefully
   - Provides detailed logging

3. **Issue Generation**:
   - Creates standardized GitHub issues for each folder
   - Includes file counts and task checklists
   - Provides clear formatting instructions

Usage examples:

```bash
# List all top-level folders with C++ files
./format_folders.py --root /path/to/fprime

# Generate GitHub issues for all folders
./format_folders.py --root /path/to/fprime --create-issues

# Check formatting for a specific folder
./format_folders.py --root /path/to/fprime --check --format Fw

# Actually format a specific folder
./format_folders.py --root /path/to/fprime --format Fw
```

Each generated issue includes:
- Folder statistics (number of files by type)
- Task checklist
- Notes about clang-format version and style
- Build testing reminder

kevin-f-ortega, 2025-05-09:
@LeStarch @thomas-bc Could formatting the C++ codebase be part of the next release?

thomas-bc, 2025-05-21:
We could use the PyPi distribution https://pypi.org/project/clang-format/ to get around the clang-format versioning issue

closed 2025-08-06 by LeStarch