nasa/fprime #1986: Implement ComBuffer DataProduct writer

timcanham, 2023-04-25:
| | |
|:---|:---|
|**_F´ Version_**|3.x |
|**_Affected Component_**| New Component |
---
## Feature Description

This component would be an alternative to `Svc/ComLogger`. It would allow `Fw/Com/ComBuffer` data to be stored as [Data Products](https://github.com/nasa/fprime/discussions/1154) as opposed to raw named files. This allows telemetry to be written to storage for the times there is not radio coverage and allows subsequent downloads to get the data products to the ground.

## Rationale

`Svc/ComLogger` works just fine, but requires managing files by name, which can complicate operations. The new [Data Product](https://github.com/nasa/fprime/discussions/1154) approach will allow the `ComBuffer` histories to be downloaded as part of a prioritized data product management approach that doesn't require explicit file names.

timcanham, 2023-04-25:
Proposed requirements:

1) The component shall request buffers of a certain size specified during initialization. (Typically connect to `Svc/BufferManager`)
1) The component shall fill the buffer with `Fw::ComBuffer` copies until full.
1) The component will send the full buffer out the write port and request another
1) The component will have a command to start and stop logging
1) The default priority of the data product will be specified as an FPP constant outside the component directory so it can be configured by user projects
1) The component will have a command to modify the data product priority dynamically (allows prioritization of telemetry for certain critical times)

matt392code, 2025-01-07:
[com-buffer-dp-writer.txt](https://github.com/user-attachments/files/18328707/com-buffer-dp-writer.txt)
[com-buffer-dp-writer-impl.txt](https://github.com/user-attachments/files/18328712/com-buffer-dp-writer-impl.txt)
Proposed complete F' component that meets all the specified requirements for storing ComBuffer data as Data Products. The key aspects:

1. Component Structure:
- The component is placed in the `Svc` namespace alongside `ComLogger`
- Uses standard F' ports for commands, events, telemetry, and time
- Defines specific ports for ComBuffer input, buffer management, and Data Product writing

2. Requirements Implementation:
- Requirement 1: Uses `BufferManager` interface via `bufferAllocate` and `bufferDeallocate` ports
- Requirement 2: Maintains a current buffer and fills it with ComBuffer data until full
- Requirement 3: Writes full buffers as Data Products and requests new ones
- Requirement 4: Implements START/STOP commands for logging control
- Requirement 5: Uses an external constant `ComBufferDpWriterPriority` for default priority
- Requirement 6: Implements SET_PRIORITY command for dynamic priority adjustment

3. Key Features:
- Buffer management with proper allocation/deallocation
- Telemetry tracking of logging state, priority, and statistics
- Event logging for important state changes and errors
- Parameter for configurable buffer size
- Efficient buffer filling with minimal copying

4. Safety Features:
- Null pointer checks
- Buffer boundary checks
- Proper resource cleanup
- Warning events for error conditions

To use this component:

1. Add to your topology:
```fpp
instance comBufferDpWriter: Svc.ComBufferDpWriter base id 0x1000 \
  queue size 100 \
  stack size 4096 \
  priority 100
```

2. Make connections:
```fpp
connections ComBufferDpWriter {
  # Connect to BufferManager
  comBufferDpWriter.bufferAllocate -> bufferManager.bufferGetCallee
  comBufferDpWriter.bufferDeallocate -> bufferManager.bufferSendIn

  # Connect to DataProduct manager
  comBufferDpWriter.dpWrite -> dpManager.dpRecv

  # Connect telemetry source
  yourComponent.comOut -> comBufferDpWriter.comIn
}
```

3. Configure parameters:
```fpp
initialize comBufferDpWriter {
  comBufferDpWriter.BUFFER_SIZE: 16384
}
```

chris-ballard-sys, 2026-09-17:
@timcanham per our discussion, I am recommending auto-closing and reopening new data products based on the following configurable limits, whichever is reached first: number of writes, time open, file size. Also we discussed having a command for the ground to close the open data products immediately and open new ones, so that for instance the command can be sent at the start of a comm pass so the ground can theoretically receive the event/telemetry data produced off-pass.

thomas-bc-autowot, 2026-10-01:
@thomas-bc @LeStarch this issue is believed to be closed by #6075 (merged 2026-09-30 into `devel`), but it is still open. Please close it if that's correct.
Unsure because: #6075 adds `Svc/ComLoggerDp` covering the original requirements, but the 2026-09-17 comment's proposed auto-close limits (writes/time/size) and ground-commanded rollover do not appear to be implemented.

thomas-bc, 2026-10-01:
Implemented as requested - if additional features are requested, please open a new ticket.

closed 2026-10-01 by thomas-bc