nasa/fprime #1988: Add DP interface to PrmDb

timcanham, 2023-04-25:
| | |
|:---|:---|
|**_F´ Version_**| 3.x |
|**_Affected Component_**| `Svc/PrmDb` |
---
## Feature Description

Add upcoming [Data Product](https://github.com/nasa/fprime/discussions/1154) interface and a command create a DP to PrmDb.

## Rationale

This would allow the ground system to see what the current values for onboard parameters are.

matt392code, 2025-01-08:
Proposed implementation approach:
[prmdb-dp.txt](https://github.com/user-attachments/files/18342325/prmdb-dp.txt)
This implementation:
1. Adds a new Data Product port type in the Fw module
2. Extends the PrmDb component with:
   - A Data Product output port
   - A CREATE_PRM_DP command to create DPs for parameters
   - Events for DP creation success/failure
3. Implements the command handler and parameter update logic

To use this feature:

1. The ground system can send the CREATE_PRM_DP command for parameters it wants to monitor
2. When parameters are updated (via setPrm), the component automatically updates the corresponding Data Products
3. The ground system receives parameter values through the Data Product interface

Key considerations:
- Used existing ParamEntry type for consistency
- Added parameter-to-DP mapping tracking
- Included appropriate error handling and event logging
- Maintained backward compatibility with existing PrmDb functionality

LeStarch, 2026-09-11:
CCB: This is to save the current in-RAM DB as a Data Product.

LeStarch, 2026-11-05:
Sending this back to CCB.  My proposal is that we could just use a more generic file-to-dp feature.  Since PrmDb is a file we could just take the existing on-disk file and send it down as a data product.

That way this procedure has exact semantics: the data product represents exactly the parameters that will be reloaded if a reboot happened now.

jayzuccarelli, 2026-11-15:
Is this one open for an outside contributor? If so, I'd take it: a `CREATE_PRM_DP` command on PrmDb that writes the current in-RAM parameter database as a Data Product, with unit tests and an SDD update. Happy to follow whatever record layout you prefer.