nasa/fprime #1991: When Changin FPP Annotations Only, Dictionary Does Not Rebuild

LeStarch, 2023-04-27:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

When changing only an annotation in an FPP model, the dictionary may not rebuild to cache the changes (GDS dictionary tab).

jwest115, 2023-05-08:
I did the below steps to try to reproduce the issue:
- Opened deployment (in my case, I used the HelloWorld example)
- Started F Prime GDS
- Edited command annotation in HelloWorld.fpp
- Ran fprime-build
    - After building, I see the change in the dictionary XML that exists in build-artifacts/dict
- Check GDS dictionary tab, dictionary content did not change to reflect updated description for command
- Shutdown F Prime GDS
- Re-started F Prime GDS
- Confirm dictionary changed to reflect command description update

LeStarch, 2023-05-08:
I would argue that is expected.  The dictionary rebuilt, which is the key piece.

LeStarch, 2023-05-08:
@jwest115 if you agree that this issue was not reproducible, would you close this issue?

jwest115, 2023-05-08:
Agreed!

closed 2023-05-08 by jwest115