nasa/fprime #2000: Add Build Repository With Latest CI Action

LeStarch, 2023-05-03:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Add a reusable action for Github CI that will build a repository with a parameterized version, defaulting to `devel`.  That way running CI on a specific repo is straight forward.

Steps:
- [ ] Clone repo (w/ submodule update)
- [ ] Update fprime version to devel
- [ ] Generate, build, unit-test, and integration test

thomas-bc, 2023-05-19:
v0 can be found here: https://github.com/fprime-community/project-builder

This is used in this sample-project: https://github.com/thomas-bc/sample-fprime-project, see the [workflow files](https://github.com/thomas-bc/sample-fprime-project/tree/main/.github/workflows) and the [Actions runs](https://github.com/thomas-bc/sample-fprime-project/actions).

I created both an Action and a Reusable Workflow. My reasoning is that a Reusable Workflow is a little more easy to interact with and debug. An Action will throw **all** the output (so generate, build, check etc...) in a single step, which is a **lot** of text and harder to debug. However a Reusable Workflow can only be used as is, so if projects want to just build their project and go on with other tasks to run in the CI, they can use the Action.

I would recommend projects to start early with the Reusable Workflow, and if they want to get more involved in the CI, they can use the Action instead, and customize/run additional things.

thomas-bc, 2023-05-19:
I was going to say "how do you want to go about Integration Testing", and I just thought, I'm going to add an optional last step to the Action. 
The users should be able to optionally specify a shell script to execute, post-build. This will allow for integration testing, or any other small things they want to be running.

thomas-bc, 2023-05-24:
- [ ] remove postbuild_script 
- [ ] either run UT or build in the Action, not both
- [ ] add URL config for being able to build a foreign repository

thomas-bc, 2023-05-26:
Split into 2 objectives:
- reusable workflow specifically for F´ core CI (building external repositories): see #2049 
- reusable workflow + Action for community usage: https://github.com/fprime-community/project-builder

thomas-bc, 2023-05-30:
https://github.com/fprime-community/project-builder should be ready for community usage

closed 2023-06-08 by thomas-bc