nasa/fprime #2009: F´ Arduino: Build System Revitalization

LeStarch, 2023-05-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

We need to revitalize F´ Arduino Build System as it is a good spring-board for ultra-small baremetal work.

capsulecorplab, 2023-05-17:
What is the scope of work needed to realize this effort?

LeStarch, 2023-05-17:
@capsulecorplab the work breaks down as:
- [x] Update wrapper to infer build settings from Arduino. [Completed In Branch](https://github.com/SterlingPeet/arduino-cli-cmake-wrapper/tree/update/refactor)
- [x] Fix the CMake integration to use wrapper.  [Completed In Branch](https://github.com/SterlingPeet/fprime-arduino/tree/mbed-work)
- [ ] Build test deployment leveraging cmake integration.  In progress.

capsulecorplab, 2023-06-24:
@LeStarch Cool~ Let me know how I can help with the test deployment. I recently updated my IIaC workspace with fprime v3.2.0 python dependencies and am also working on being able to run the system reference
- https://github.com/capsulecorplab/fprime-workspace-image
- https://github.com/capsulecorplab/fprime-workspace-image/tree/install-opencv-as-part-of-fprime-system-reference
- https://github.com/users/capsulecorplab/packages/container/package/fprime-workspace-image
- https://github.com/users/capsulecorplab/packages/container/fprime-workspace-image/103290523?tag=install-opencv-as-part-of-fprime-system-reference

closed 2023-08-16 by LeStarch