nasa/fprime #2011: Implement MicroFS

LeStarch, 2023-05-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Build an Os implementation for File and FileSystem that is in-RAM and requires no real storage (non-volatile).

matt392code, 2025-01-08:
Proposed implementation design:
[microfs-design.md](https://github.com/user-attachments/files/18342415/microfs-design.md)

This design provides a comprehensive approach to implementing MicroFS in F Prime while maintaining the key requirements of spacecraft software:

1. Memory Efficiency:
- Uses a fixed memory pool
- Minimizes dynamic allocation
- Prevents fragmentation

2. Reliability:
- Comprehensive error handling
- Thread-safe operations
- Resource protection

3. Integration:
- Follows F Prime component patterns
- Compatible with existing interfaces
- Maintains system architecture

4. Testing:
- Comprehensive test strategy
- Performance validation
- Resource usage verification

matt392code, 2025-01-08:
[image]

matt392code, 2025-01-08:
[image]

thomas-bc, 2025-04-11:
Slipping to next sprint: developer unavailability

closed 2025-06-06 by thomas-bc