nasa/fprime #2020: Telemetry channels not added to RefPackets.xml upon build

NoahBraasch, 2023-05-12:
| | |
|:---|:---|
|**_F´ Version_**|3.2.0 |
|**_Affected Component_**|   |
---
## Problem Description

A description of the problem with sufficient detail to understand the issue.

## How to Reproduce
Follow math component tutorial up to step 6.3 sans building unit tests. 
The build will fail when packetizing custom telemetry channels
`Packet SigGen4 size 73/111
Packetizing SigGen5 (19)
Packet SigGen5 size 73/111
Packetizing TypeDemo (20)
Packet TypeDemo size 81/111
Channel mathSender.VAL1 (3840) not packetized or ignored.
gmake[3]: *** [Ref/Top/CMakeFiles/Ref_Top.dir/build.make:220: Ref/Top/RefPacketsAc.cpp] Error 255
gmake[2]: *** [CMakeFiles/Makefile2:9410: Ref/Top/CMakeFiles/Ref_Top.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:2394: CMakeFiles/Ref.dir/rule] Error 2
gmake: *** [Makefile:183: Ref] Error 2
Channel mathSender.OP (3841) not packetized or ignored.
Channel mathSender.VAL2 (3842) not packetized or ignored.
Channel mathSender.RESULT (3843) not packetized or ignored.
Channel mathReceiver.OPERATION (9984) not packetized or ignored.
Channel mathReceiver.FACTOR (9985) not packetized or ignored.
Packet XML parsing error: Channels missing from packets
[ERROR] CMake erred with return code 2`

This is because the channels are not being added to RefPackets.xml

Manually adding these channels fixes the issue.

## Expected Behavior

A description of the expected behavior.

LeStarch, 2023-05-12:
We need to add this to the Math Component Tutorial!  Thanks for reporting the issue.

NoahBraasch, 2023-05-12:
> We need to add this to the Math Component Tutorial! Thanks for reporting the issue.

Happy to help. Is the pages repo something that anyone can contribute to? If so, id be happy to help with the revision!

LeStarch, 2023-05-13:
All our pages are written as markdown under the `docs` folder.  Here is a link to that tutorial if you'd like to help out: https://github.com/nasa/fprime/blob/master/docs/Tutorials/MathComponent/Tutorial.md