nasa/fprime #2030: EPIC: Walk Through Documentation and Do an Editorial Pass

LeStarch, 2023-05-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Documentation version-2!

thomas-bc, 2023-05-17:
For reference, I was at a conference and someone** presented about how they have improved their documentation using the [Diataxis method](https://diataxis.fr/). The TL;DR is there's 4 types of documentation - tutorials, how-to guides, technical reference and design explanation. And sticking to that doc "architecture" helps consuming it efficiently.
I think we're not too far from there, we would just need to reorganize things a little and clearly mark the distinctions.

**That someone was really Project Jupyter, a fairly large open-source project and I trust them to implement good practice. Example docs that followed the Diataxis method: https://jupyterhub.readthedocs.io/en/stable/

thomas-bc, 2023-05-30:
[Architecture Overview PDF](https://github.com/nasa/fprime/tree/devel/docs/Architecture) should be replaced with one of the newer versions we have that we use to present at workshops

https://github.com/nasa/fprime/blob/4f65e1ef3fa3aa44e132e6d9e7b885e840162630/docs/index.md?plain=1#L27

thomas-bc, 2023-05-30:
Landing page of the F´ website (https://nasa.github.io/fprime/) could also be reorganized to be easier to navigate for new users. Install instructions link could be changed to link to HelloWorld tutorial, links to `Ref/` and `RPI/` should be scrubbed and replaced by LedBlinker et al. 
It also may not be necessary to link to all versions of the documentation on the landing page, rather just point at the latest, and give a link that points to a list of other versions if necessary.

LeStarch, 2024-03-27:
@cgcruz0304 you basically did this and yielded recommendations.  I am going to close this issue.

closed 2024-03-27 by LeStarch