nasa/fprime #2036: Fw_Tlm fails to build when cross-compiling

scoliono, 2023-05-18:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**| Fw::Tlm  |
---
## Problem Description

Hi, I'm with UCLA's ELFIN CubeSat team. We recently attended the F' tutorial session at the Cal Poly CubeSat conference and are exploring it as a possible option for future missions' flight software.

I'm trying to cross-compile for FreeRTOS running on a SmartFusion2 (Cortex-M3). I noticed there was a community FreeRTOS port and tried to model my CMake setup around that. My team isn't using GitHub but my relevant modifications are here: https://gist.github.com/scoliono/11443537b0b9185c0a6d5a813e77e206.

However, the build fails, and it doesn't appear to be due to any of my changes:

```
Consolidate compiler generated dependencies of target Fw_Tlm
[ 11%] Building CXX object F-Prime/Fw/Tlm/CMakeFiles/Fw_Tlm.dir/TlmPacket.cpp.obj
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp: In member function 'virtual Fw::SerializeStatus Fw::TlmPacket::serialize(Fw::SerializeBufferBase&) const':
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:134:67: error: call of overloaded 'serialize(const NATIVE_UINT_TYPE&)' is ambiguous
  134 |         SerializeStatus stat = buffer.serialize(this->m_numEntries);
      |                                                                   ^
In file included from /home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:7:
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:50:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(U8)'
   50 |             SerializeStatus serialize(U8 val); //!< serialize 8-bit unsigned int
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:51:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(I8)'
   51 |             SerializeStatus serialize(I8 val); //!< serialize 8-bit signed int
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:54:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(U16)'
   54 |             SerializeStatus serialize(U16 val); //!< serialize 16-bit unsigned int
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:55:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(I16)'
   55 |             SerializeStatus serialize(I16 val); //!< serialize 16-bit signed int
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:58:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(U32)'
   58 |             SerializeStatus serialize(U32 val); //!< serialize 32-bit unsigned int
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:59:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(I32)'
   59 |             SerializeStatus serialize(I32 val); //!< serialize 32-bit signed int
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:65:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(F32)'
   65 |             SerializeStatus serialize(F32 val); //!< serialize 32-bit floating point
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:69:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::serialize(bool)'
   69 |             SerializeStatus serialize(bool val); //!< serialize boolean
      |                             ^~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp: In member function 'virtual Fw::SerializeStatus Fw::TlmPacket::deserialize(Fw::SerializeBufferBase&)':
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:145:69: error: no matching function for call to 'deserialize(NATIVE_UINT_TYPE&)'
  145 |         SerializeStatus stat = buffer.deserialize(this->m_numEntries);
      |                                                                     ^
In file included from /home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:7:
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:81:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::deserialize(U8&)' (near match)
   81 |             SerializeStatus deserialize(U8 &val); //!< deserialize 8-bit unsigned int
      |                             ^~~~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:81:29: note:   conversion of argument 1 would be ill-formed:
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:145:57: error: cannot bind non-const lvalue reference of type 'U8&' {aka 'unsigned char&'} to an rvalue of type 'U8' {aka 'unsigned char'}
  145 |         SerializeStatus stat = buffer.deserialize(this->m_numEntries);
      |                                                   ~~~~~~^~~~~~~~~~~~
In file included from /home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:7:
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:82:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::deserialize(I8&)' (near match)
   82 |             SerializeStatus deserialize(I8 &val); //!< deserialize 8-bit signed int
      |                             ^~~~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:82:29: note:   conversion of argument 1 would be ill-formed:
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:145:57: error: cannot bind non-const lvalue reference of type 'I8&' {aka 'signed char&'} to an rvalue of type 'I8' {aka 'signed char'}
  145 |         SerializeStatus stat = buffer.deserialize(this->m_numEntries);
      |                                                   ~~~~~~^~~~~~~~~~~~
In file included from /home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:7:
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:85:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::deserialize(U16&)' (near match)
   85 |             SerializeStatus deserialize(U16 &val); //!< deserialize 16-bit unsigned int
      |                             ^~~~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:85:29: note:   conversion of argument 1 would be ill-formed:
[... 44 lines cut ...]
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:103:29: note: candidate: 'Fw::SerializeStatus Fw::SerializeBufferBase::deserialize(void*&)' (near match)
  103 |             SerializeStatus deserialize(void*& val); //!< deserialize point value (careful, pointer value only, not contents)
      |                             ^~~~~~~~~~~
/home/elfin/Duchess/fprime/Fw/Types/Serializable.hpp:103:29: note:   conversion of argument 1 would be ill-formed:
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:145:57: error: invalid conversion from 'NATIVE_UINT_TYPE' {aka 'unsigned int'} to 'void*' [-fpermissive]
  145 |         SerializeStatus stat = buffer.deserialize(this->m_numEntries);
      |                                                   ~~~~~~^~~~~~~~~~~~
      |                                                         |
      |                                                         NATIVE_UINT_TYPE {aka unsigned int}
/home/elfin/Duchess/fprime/Fw/Tlm/TlmPacket.cpp:145:57: error: cannot bind rvalue '(void*)((Fw::TlmPacket*)this)->Fw::TlmPacket::m_numEntries' to 'void*&'
gmake[2]: *** [F-Prime/Fw/Tlm/CMakeFiles/Fw_Tlm.dir/build.make:120: F-Prime/Fw/Tlm/CMakeFiles/Fw_Tlm.dir/TlmPacket.cpp.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2205: F-Prime/Fw/Tlm/CMakeFiles/Fw_Tlm.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
[ERROR] CMake erred with return code 2
```

## How to Reproduce

1. Use the files from the Gist
2. fprime-util generate smartfusion2
3. fprime-util build -j4 smartfusion2

## Expected Behavior

The build should succeed as it does when not cross-compiling.

LeStarch, 2023-05-19:
@scoliono this is an issue in the BSP layer of your project combined with certain compiler setups.  In short, within `PlatformTypes.h` for your platform, you must set the platform int and uint definitions to a sized integer.  I have linked to the platform types shipped with F´: 

https://github.com/nasa/fprime/blob/4f65e1ef3fa3aa44e132e6d9e7b885e840162630/cmake/platform/types/PlatformTypes.h#L41-L45

In your platfom (smartfusion2) that should read:

```
typedef int32_t PlatformIntType;
#define PRI_PlatformIntType "d"

typedef uint32_t PlatformUIntType;
#define PRI_PlatformUIntType "u"
```
> This assumes your bus-width is 32bit.  Adjust accordingly.

LeStarch, 2023-05-19:
Since this isn't something F´  can fix as the fix goes inside your project, I'll eventually close this issue.  However, for some time I will leave it open in-case there is follow-up discussion needed!

Good luck!

scoliono, 2023-05-25:
Thanks for your help! I tried your changes and ran into some more errors, but eventually got it to compile. In case anyone else has the same issue, it turns out my compiler (arm-none-eabi-g++) set the `__(u)int32_t` macro to a `long (unsigned) int`, requiring the PRI_Platform(U)IntType macros to be changed as well. On top of that, sys/types.h needed to be included before inttypes.h so the macros like PRId64, PRIu64 would be correctly defined by newlib.

closed 2023-05-25 by scoliono