nasa/fprime #2059: Add Visulazation Script

LeStarch, 2023-05-31:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Runs visualizer setup commands and visualizer.

thomas-bc, 2023-06-22:
Opened https://github.com/fprime-community/fprime-tools/issues/33 - it still depends on the other linked PRs to be merged/handled, and FPP-Layout to be distributed as a pip package, by itself (or better within FPP?)

thomas-bc, 2023-07-06:
@LeStarch @bocchino 
All the pieces are there, they just need reviewing and releasing:
- https://github.com/fprime-community/fprime-layout/pull/1
   Very minor change to FPL install script
- https://github.com/fprime-community/fpp/pull/272
   Need to release FPP and publish to PyPi. Allows for `fprime-layout` tools to be distributed with PyPi
- https://github.com/fprime-community/fprime-visual/pull/1
   Need to release and publish to PyPi.
- https://github.com/fprime-community/fprime-tools/pull/147: 
  `fprime-util visualize` piecing it all together.

thomas-bc, 2023-08-07:
Released as of v3.3.0

closed 2023-08-07 by thomas-bc