nasa/fprime #2073: Reconsider Polling Buffer in Deframer Code

LeStarch, 2023-06-08:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Deframer has poll and `schedIn` functionality.  Does it need this when the driver itself could be polled (e.g. on a rate group) in the case that the driver does not have a thread.

matt392code, 2025-01-08:
Proposed detailed implementation plan for removing the polling buffer functionality from the Deframer component:
[deframer-changes.txt](https://github.com/user-attachments/files/18342501/deframer-changes.txt)
Here's a breakdown of the proposed changes:

1. **Architectural Change**:
   - Remove polling functionality entirely from Deframer
   - Simplify to only support callback-based operation
   - Move polling responsibility to driver components

2. **Key Benefits**:
   - Cleaner separation of concerns
   - Simplified component interface
   - More flexible driver implementations
   - Reduced complexity in Deframer code

3. **Migration Steps**:
   - Remove `schedIn` and `poll` ports
   - Remove polling-related internal state
   - Update existing drivers to handle polling if needed
   - Update component documentation

4. **Implementation Considerations**:
   - Ensure backward compatibility path for existing systems
   - Update unit tests to remove polling scenarios
   - Verify buffer management remains correct
   - Document migration path for existing implementations

LeStarch, 2025-01-23:
@thomas-bc this will be fixed by the deframer rework, yes?  If so, would you close?

closed 2025-02-04 by LeStarch